tinygo-org/tinygo #4145: error: requires go version 1.18 through 1.21, got go1.22

lidldev, 2024-02-22:
Hey, I try to use tinygo on my Mac but I cant when I try to flash the hello world app to my Arduino it gives me this error
And the machine package does not appear at first, but tinygo versions etc. all appear.

error in vscode ->
``cmd
Could not run 'tinygo info arduino-zero': Error: Command failed: tinygo info arduino-zero requires go version 1.18 through 1.21, got go1.22 TinyGo is a Go compiler for small places. version: 0.30.0 usage: /opt/homebrew/Cellar/tinygo/0.30.0/libexec/tinygo <command> [arguments] commands: build: compile packages and dependencies run: compile and run immediately test: test packages flash: compile and flash to the device gdb: run/flash and immediately enter GDB lldb: run/flash and immediately enter LLDB monitor: open communication port env: list environment variables used during build list: run go list using the TinyGo root clean: empty cache directory (/Users/ahmetaffanebcioglu/Library/Caches/tinygo) targets: list targets info: show info for specified target version: show version help: print this help text flags: -baudrate int baudrate of serial monitor (default 115200) -cpuprofile string cpuprofile output -gc s...
``

weird...

mehmetakyurek, 2024-02-23:
You can see supported go versions here: https://tinygo.org/docs/reference/go-compat-matrix/
tinygo doesn't support go1.22 yet, so you need to install go1.20. This might be useful https://github.com/moovweb/gvm

lidldev, 2024-02-23:
> You can see supported go versions here: https://tinygo.org/docs/reference/go-compat-matrix/ tinygo doesn't support go1.22 yet, so you need to install go1.20. This might be useful https://github.com/moovweb/gvm

Teşekkür ederim!

closed 2024-02-23 by lidldev