tinygo-org/tinygo #4148: Building from docker image: "tinygo:ld.lld: error: undefined symbol"

orsinium, 2024-02-24:
`Dockerfile`:

```Dockerfile
FROM tinygo/tinygo:latest
COPY ./go.mod /app/
COPY ./echo/ /app/echo
WORKDIR /app/
RUN tinygo build -o echo.bin ./echo
```

`echo/echo.go`:

```go
package main

import (
	"net/http"
)

func main() {
	http.HandleFunc("/", nil)
}
```

`go.mod`:

```
module echo

go 1.21.3
```

Running `sudo docker build .` produces the following output:

```
 > [5/5] RUN tinygo build -o echo.bin ./echo:                                                                                                                                               
13.52 tinygo:ld.lld: error: undefined symbol: vendor/golang.org/x/sys/cpu.cpuid                                                                                                             
13.52 >>> referenced by main                                                                                                                                                                
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:(runtime.initAll)                                                             
13.52 >>> referenced by main                                                                                                                                                                
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:(runtime.initAll)
13.52 >>> referenced by main
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:(runtime.initAll)
13.52 >>> referenced 1 more times
13.52 
13.52 tinygo:ld.lld: error: undefined symbol: vendor/golang.org/x/sys/cpu.xgetbv
13.52 >>> referenced by main
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:(runtime.initAll)
13.52 
13.52 tinygo:ld.lld: error: undefined symbol: crypto/internal/edwards25519/field.feSquare
13.52 >>> referenced by fe.go:309 (/usr/local/go/src/crypto/internal/edwards25519/field/fe.go:309)
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:((*crypto/internal/edwards25519/field.Element).Square)
13.52 
13.52 tinygo:ld.lld: error: undefined symbol: crypto/internal/edwards25519/field.feMul
13.52 >>> referenced by fe.go:303 (/usr/local/go/src/crypto/internal/edwards25519/field/fe.go:303)
13.52 >>>               /home/tinygo/.cache/tinygo/thinlto/llvmcache-1ACD7D0E614F979B6E33B7C6AD6247EE23EFB5D6:((*crypto/internal/edwards25519/field.Element).Multiply)
13.52 failed to run tool: ld.lld
13.65 error: failed to link /tmp/tinygo1036176333/main: exit status 1
```

aykevl, 2024-03-13:
This may have been fixed with https://github.com/tinygo-org/tinygo/pull/4116, can you test again with the latest release?

orsinium, 2024-03-21:
It works! Thank you

closed 2024-03-21 by orsinium