tinygo-org/tinygo #4151: code runs as expected but kernel shows "no enough memory for the allocation"

laurento, 2024-02-25:
I compiled my small tool using tinygo because I'm targeting a Raspberry Pi first generation with 256MB of RAM, so every byte counts :) It works just fine. However I just noticed that it generates the following kernel error every time it runs.

`[33719.335172] __vm_enough_memory: pid: 7816, comm: mystuff, no enough memory for the allocation`

If I compile it using `GOOS=linux GOARCH=arm GOARM=6 tinygo build mystuff.go`. If I run the same code but compiled it using `go` the kernel doesn't seem to complain.

For my tests I'm using:
Linux raspi 6.1.0-17-rpi #1 Debian 6.1.69-1 (2023-12-30) armv6l GNU/Linux
tinygo version 0.30.0 linux/amd64 (using go version go1.21.7 and LLVM version 16.0.1)

peritonow, 2024-04-26:
I am new to tinygo as well, but there is no much to go on with the information you provided.

From my understanding this go code is translate to c and recompiled with "gcc" ... 
https://forum.arduino.cc/t/not-enough-memory-coding-troubleshooting/417437 and the error is from c compiler.

so this might be a bug withing c library and nothing to do with tinygo... also might be related to go stack escape that embedded programming might not tolerate and you need to find out the lines within your code. Hard to say.

aykevl, 2024-04-27:
It's probably this code:

https://github.com/tinygo-org/tinygo/blob/6384ecace093df2d0b93915886954abfc4ecfe01/src/runtime/runtime_unix.go#L220-L237

I'm surprised it prints a warning. Maybe the kernel really doesn't like an 1GB allocation? (It should just return an error if it is too big).

> From my understanding this go code is translate to c and recompiled with "gcc" ...
> https://forum.arduino.cc/t/not-enough-memory-coding-troubleshooting/417437 and the error is from c compiler.

It is not, we use LLVM as a compiler toolchain (and don't use GCC anywhere in the process).

laurento, 2024-04-27:
Perhaps this kernel patch is relevant https://lore.kernel.org/linux-mm/18dc0807-5596-3aad-350d-3673a033bca8@redhat.com/T/

deadprogram, 2024-12-07:
Is there some action required on this issue, or can it be closed?

aykevl, 2025-09-30:
Closing, since I'm not seeing any activity on it. Please comment if this is still a problem.

closed 2025-09-30 by aykevl