tinygo-org/tinygo #4163: build  '-target=wasm-unknown'   pure wasm fail

ltp456, 2024-02-28:
tinygo version:  tinygo version 0.31.0 darwin/arm64 (using go version go1.22.0 and LLVM version 17.0.6)
os system:   mac os 14.3
build cmd:
`tinygo build -o ecdsa.wasm  -target=wasm-unknown  ./main.go`

error log:
``/Users/red/lworkspace/dfinity/tinygo/src/syscall/syscall_unix.go:10:7: undefined: RawSockaddrInet4
/Users/red/lworkspace/dfinity/tinygo/src/syscall/syscall_unix.go:17:9: undefined: RawSockaddrInet6
/Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:28:14: undefined: EINTR
/Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:28:28: undefined: EMFILE
/Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:32:14: undefined: EAGAIN
/Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:32:29: undefined: EWOULDBLOCK
/Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:32:49: undefined: ETIMEDOUT
/Users/red/lworkspace/dfinity/tinygo/src/syscall/file_hosted.go:14:14: undefined: getErrno
/Users/red/lworkspace/dfinity/tinygo/src/syscall/file_hosted.go:18:14: undefined: EINVAL`

aykevl, 2024-03-13:
You are mixing a number of different bugs in a single report.

> `/Users/red/lworkspace/dfinity/tinygo/src/syscall/syscall_unix.go:10:7: undefined: RawSockaddrInet4 /Users/red/lworkspace/dfinity/tinygo/src/syscall/syscall_unix.go:17:9: undefined: RawSockaddrInet6 /Users/red/lworkspace/dfinity/tinygo/src/syscall/errno.go:28:14: undefined: EINTR`

Probably fixed in https://github.com/tinygo-org/tinygo/pull/4168. Please test with tinygo 0.31.2.

> `clang-17: error: no such file or directory: '/Users/red/lworkspace/dfinity/tinygo/lib/compiler-rt-builtins/addvsi3.c' clang-17: error: no input files clang-17: error: no such file or directory: '/Users/red/lworkspace/dfinity/tinygo/lib/compiler-rt-builtins/absvdi2.c' clang-17: error: no input files `

Not a bug, you need to run `make llvm-source` (see https://tinygo.org/docs/guides/build/additional-requirements/).

> then try download `compiler-rt-builtins`(https://github.com/ilobilo/compiler-rt-builtins) to tinygo/lib/compiler-rt-builtins dirs

Don't do this, in fact I recommend undoing this so they don't mess with your tinygo build. The correct fix is `make llvm-source`.

> `wasm-ld: error: lto.tmp: undefined symbol: memset `

See https://github.com/tinygo-org/tinygo/issues/4174

deadprogram, 2024-12-06:
Closing since seems like nothing further needed. Thanks!

closed 2024-12-06 by deadprogram