tinygo-org/tinygo #4170: nrf: writes via `machine.Flash` cause assertion violations with SoftDevice enabled

bgould, 2024-03-03:
When performing writes via the NVMC registers, the SoftDevice generates an assertion violation if enabled.

In the nRF SDK, there are 2 "backends" for flash storage, one via the NVMC controller (as in `machine.Flash`) and another that designed for compatibility with SoftDevice enabled: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fstruct_sdk%2Fstruct%2Fsdk_nrf5_latest.html

It would be nice to be able to support both backends;  It could be the case that the implementation of the SoftDevice backend belongs in tinygo.org/x/bluetooth (or a subpackage).  In that case, it might be beneficial to refactor `machine.Flash` to allow for specifying a "backend" other than the NVMC implementation.

aykevl, 2024-03-03:
I was afraid we'd run into this.

Perhaps the least complicated way to solve this is by doing it in the same way I've solved `wfi` before:
https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_nrf_softdevice.go

deadprogram, 2025-01-12:
@bgould @aykevl if https://github.com/tinygo-org/tinygo/pull/4692 is good, I can work on fix for this one as well.

davecheney, 2026-10-27:
Thanks for reporting this. Flash operations with the SoftDevice enabled are now supported by #5090, with a follow-up correction to writes in #5122.

Closing this as resolved for now. Please feel free to reopen if either fix is incomplete or you still encounter assertion failures.

closed 2026-10-27 by davecheney