tinygo-org/tinygo #4177: No support for new ARM softfloat/hardfloat distinction added in Go 1.22

MDr164, 2024-03-06:
When trying to compile a binary with the new ARM subtargets, e.g. `GOARM=6,softfloat`, tinygo bails out. The support for ARM added in #2260 would need to be extended for that.

aykevl, 2024-03-12:
I believe TinyGo already uses softfloat with `GOARM=6` because not all ARMv6 chips have an FPU.
Is this not the case for you?
(Can you explain what system you'd like to compile for?)

MDr164, 2024-03-13:
Actually I would need both ARMv6 and ARMv7 with softfloat support for my usecase and it's more about supporting the same naming scheme that upstream Go uses since 1.22. I can have a try at implementing this as I got the chips here that require softfloat due to a lack of an FPU. The systems I'm looking to compile for are the Aspeed BMC chips. Both the 2500 and 2600 are missing the FPU.

aykevl, 2024-03-13:
> and ARMv7 with softfloat support

Right, I wasn't aware these cores existed. Thanks for clearing that up!
(You'd still be able to compile with `GOARM=6` for compatibility but it would be slower).

MDr164, 2024-03-13:
Right, that's also why this isn't anything high prio or blocking but might as well have it in to make use of the v7 features without an FPU. Still need to figure out which LLVM flags I'll need to set but otherwise the linked PR should already handle what I'm trying to achieve.

aykevl, 2024-03-13:
> [...] but otherwise the linked PR should already handle what I'm trying to achieve.

Probably not, you need the correct LLVM feature flags. Those define the available instructions (and a few other things) that LLVM can use in the backend. Right now they are identical for ARMv7 softfloat and appear to include floating point instructions.

MDr164, 2024-03-13:
That's why I wrote that I still need to figure out the flags :sweat_smile:.
I left the identical for the draft just so I can post the general idea of what I'd change :)

davecheney, 2026-10-27:
Thanks for reporting this. ARM softfloat support was added in #4374. I verified both requested configurations with the current TinyGo tree using this program:

```go
package main

//go:noinline
func divide(a, b float64) float64 {
	return a / b
}

func main() {
	println(divide(7, 2))
}
```

```sh
GOOS=linux GOARCH=arm GOARM=6,softfloat tinygo build -o armv6 main.go
GOOS=linux GOARCH=arm GOARM=7,softfloat tinygo build -o armv7 main.go
```

Both builds completed successfully. Closing as resolved. Please feel free to reopen if you encounter any remaining problems with these configurations.

closed 2026-10-27 by davecheney