tinygo-org/tinygo #4178: Bus Error on Raspberry Pi 4 B When Running Linked List Append

MjMoshiri, 2024-03-08:
## Description
When attempting to run a simple linked list implementation that appends nodes to the list in a loop, I encounter a "bus error" on a Raspberry Pi 4 Model B. Interestingly enough when I compile the same to some other targets (e.g. WASI) it works.

### Environment
Hardware: Raspberry Pi 4 Model B
Operating System: Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
TinyGo Version: tinygo version 0.31.1 linux/arm64 (using go version go1.22.0 and LLVM version 17.0.1)

### Affected Code:

```go
package main

import (
	"time"
)

type Node struct {
	data int
	next *Node
}

func append(head *Node, data int) *Node {
	newNode := &Node{data: data}

	if head == nil {
		head = newNode
	} else {
		currentNode := head
		for currentNode.next != nil {
			currentNode = currentNode.next
		}
		currentNode.next = newNode
	}

	return head
}

func main() {
	start := time.Now()

	var head *Node
	for i := 0; i < 10000; i++ {
		head = append(head, i)
	}
	duration := time.Since(start)
	println(duration.Nanoseconds())
}
```
Compiled Using
```bash
GOARCH=arm64 GOOS=linux GOGC=off tinygo build -opt=0 -scheduler=none
```

aykevl, 2024-03-12:
Have you tried without `-opt=0`? Disabling optimizations often leads to bugs.

I might remove that flag entirely, I've seen so many bug reports where `-opt=0` was causing bugs.

deadprogram, 2024-12-06:
Closing since this question appears to have been answered. Please reopen if needed. Thank you!

closed 2024-12-06 by deadprogram