tinygo-org/tinygo #4179: Joystick USB HID device - changing identity and buttons

andrewfstratton, 2024-03-08:
Firstly it's great to have so much working already...

When I use https://pkg.go.dev/github.com/tinygo-org/tinygo/src/machine/usb/hid/joystick this works fine but has an issue with identification/description, e.g.

- Using https://hardwaretester.com/gamepad, with a Badger2040, this is identified as 'Badger 2040 (Vendor: 2e8a Product: 0003)'
- this has the following effects:
  - The Buttons are shown as B0 to B15
    - N.B. B16 (Home) isn't shown - and if used, **appears to crash**
  - There are 9 Axis shown (I haven't tested these yet...)
  
Suggestion:
- Could the USB HID Descriptor be changed (by api) to allow XBox 360 (or similar) to be selected?
  - I have tested this with an 8BitDo controller, which identifies itself as 'Xbox 360 Controller (XInput STANDARD GAMEPAD)' - this then shows the capabilities as B0 to B16 (i.e. including Home button) and Axis 0 to 3
  - I believe that UseSettings allows this - but I can't find any examples that would help me to do this and I got lost in the detail...
  
  Note: I tried adding in the below code - but this gave me buttons 0..12 and Axis 0..5 - which might work (ish) assuming the triggers are now analogue - but drops a couple more buttons - and the Identity is still Badger2040:
  
```
func init() {
	joystick.UseSettings(joystick.DefaultDefinitions(), nil, nil, XboxHIDDescriptor)
}
```

With this as well:

```
var XboxHIDDescriptor = []byte{
	0x05, 0x01, // USAGE_PAGE (Generic Desktop)
	0x09, 0x05, // USAGE (Game Pad)
	0xa1, 0x01, // COLLECTION (Application)
	0x05, 0x01, //   USAGE_PAGE (Generic Desktop)
	0x09, 0x3a, //   USAGE (Counted Buffer)
	0xa1, 0x02, //   COLLECTION (Logical)
	0x75, 0x08, //     REPORT_SIZE (8)
	0x95, 0x02, //     REPORT_COUNT (2)
	0x05, 0x01, //     USAGE_PAGE (Generic Desktop)
	0x09, 0x3f, //     USAGE (Reserved)
	0x09, 0x3b, //     USAGE (Byte Count)
	0x81, 0x01, //     INPUT (Cnst,Ary,Abs)
	0x75, 0x01, //     REPORT_SIZE (1)
	0x15, 0x00, //     LOGICAL_MINIMUM (0)
	0x25, 0x01, //     LOGICAL_MAXIMUM (1)
	0x35, 0x00, //     PHYSICAL_MINIMUM (0)
	0x45, 0x01, //     PHYSICAL_MAXIMUM (1)
	0x95, 0x04, //     REPORT_COUNT (4)
	0x05, 0x09, //     USAGE_PAGE (Button)
	0x19, 0x0c, //     USAGE_MINIMUM (Button 12)
	0x29, 0x0f, //     USAGE_MAXIMUM (Button 15)
	0x81, 0x02, //     INPUT (Data,Var,Abs)
	0x75, 0x01, //     REPORT_SIZE (1)
	0x15, 0x00, //     LOGICAL_MINIMUM (0)
	0x25, 0x01, //     LOGICAL_MAXIMUM (1)
	0x35, 0x00, //     PHYSICAL_MINIMUM (0)
	0x45, 0x01, //     PHYSICAL_MAXIMUM (1)
	0x95, 0x04, //     REPORT_COUNT (4)
	0x05, 0x09, //     USAGE_PAGE (Button)
	0x09, 0x09, //     USAGE (Button 9)
	0x09, 0x0a, //     USAGE (Button 10)
	0x09, 0x07, //     USAGE (Button 7)
	0x09, 0x08, //     USAGE (Button 8)
	0x81, 0x02, //     INPUT (Data,Var,Abs)
	0x75, 0x01, //     REPORT_SIZE (1)
	0x15, 0x00, //     LOGICAL_MINIMUM (0)
	0x25, 0x01, //     LOGICAL_MAXIMUM (1)
	0x35, 0x00, //     PHYSICAL_MINIMUM (0)
	0x45, 0x01, //     PHYSICAL_MAXIMUM (1)
	0x95, 0x03, //     REPORT_COUNT (3)
	0x05, 0x09, //     USAGE_PAGE (Button)
	0x09, 0x05, //     USAGE (Button 5)
	0x09, 0x06, //     USAGE (Button 6)
	0x09, 0x0b, //     USAGE (Button 11)
	0x81, 0x02, //     INPUT (Data,Var,Abs)
	0x75, 0x01, //     REPORT_SIZE (1)
	0x95, 0x01, //     REPORT_COUNT (1)
	0x81, 0x01, //     INPUT (Cnst,Ary,Abs)
	0x75, 0x01, //     REPORT_SIZE (1)
	0x15, 0x00, //     LOGICAL_MINIMUM (0)
	0x25, 0x01, //     LOGICAL_MAXIMUM (1)
	0x35, 0x00, //     PHYSICAL_MINIMUM (0)
	0x45, 0x01, //     PHYSICAL_MAXIMUM (1)
	0x95, 0x04, //     REPORT_COUNT (4)
	0x05, 0x09, //     USAGE_PAGE (Button)
	0x19, 0x01, //     USAGE_MINIMUM (Button 1)
	0x29, 0x04, //     USAGE_MAXIMUM (Button 4)
	0x81, 0x02, //     INPUT (Data,Var,Abs)
	0x75, 0x08, //     REPORT_SIZE (8)
	0x15, 0x00, //     LOGICAL_MINIMUM (0)
	0x26, 0xff, 0x00, //     LOGICAL_MAXIMUM (255)
	0x35, 0x00, //     PHYSICAL_MINIMUM (0)
	0x46, 0xff, 0x00, //     PHYSICAL_MAXIMUM (255)
	0x95, 0x02, //     REPORT_COUNT (2)
	0x05, 0x01, //     USAGE_PAGE (Generic Desktop)
	0x09, 0x32, //     USAGE (Z)
	0x09, 0x35, //     USAGE (Rz)
	0x81, 0x02, //     INPUT (Data,Var,Abs)
	0x75, 0x10, //     REPORT_SIZE (16)
	0x16, 0x00, 0x80, //     LOGICAL_MINIMUM (-32768)
	0x26, 0xff, 0x7f, //     LOGICAL_MAXIMUM (32767)
	0x36, 0x00, 0x80, //     PHYSICAL_MINIMUM (-32768)
	0x46, 0xff, 0x7f, //     PHYSICAL_MAXIMUM (32767)
	0x05, 0x01, //     USAGE_PAGE (Generic Desktop)
	0x09, 0x01, //     USAGE (Pointer)
	0xa1, 0x00, //     COLLECTION (Physical)
	0x95, 0x02, //       REPORT_COUNT (2)
	0x05, 0x01, //       USAGE_PAGE (Generic Desktop)
	0x09, 0x30, //       USAGE (X)
	0x09, 0x31, //       USAGE (Y)
	0x81, 0x02, //       INPUT (Data,Var,Abs)
	0xc0,       //     END_COLLECTION
	0x05, 0x01, //     USAGE_PAGE (Generic Desktop)
	0x09, 0x01, //     USAGE (Pointer)
	0xa1, 0x00, //     COLLECTION (Physical)
	0x95, 0x02, //       REPORT_COUNT (2)
	0x05, 0x01, //       USAGE_PAGE (Generic Desktop)
	0x09, 0x33, //       USAGE (Rx)
	0x09, 0x34, //       USAGE (Ry)
	0x81, 0x02, //       INPUT (Data,Var,Abs)
	0xc0, //     END_COLLECTION
	0xc0, //   END_COLLECTION
	0xc0, // END_COLLECTION
}
```

sago35, 2024-03-08:
The following may be of some assistance to you.

https://github.com/SWITCHSCIENCE/ffb_steering_controller/
https://github.com/SWITCHSCIENCE/ffb_steering_controller/blob/93c76de0c67f00f1dda0c8022c9cb2f45eb603db/control/wheel.go#L17-L30

andrewfstratton, 2024-03-11:
I now no longer think the the identity needs to change - I may add a different issue later - I'm still testing the default descriptor and trying to get one working beyond the browser teste page I was using...

closed 2024-03-11 by andrewfstratton