tinygo-org/tinygo #4210: No space between hexadecimal literal and identifier

Taoja, 2024-03-28:
tinygo version 0.30.0 windows/amd64

i got the error `No space between hexadecimal literal and identifier` when i using iphone safari running the wasm_exec.js

looks like iphone6 safari not support `0xffffffffn`

aykevl, 2024-03-28:
> looks like iphone6 safari not support `0xffffffffn`

Probably, yeah.
The iPhone 6 is 9 years old by now, and is not supported by Apple anymore. I'm afraid we can't keep supporting very old devices like that with no updates. (If it were an Android device you could have installed a different browser, but Apple doesn't allow that kind of thing).
If that's a problem, you can either use an older version of TinyGo or try to revert this PR: https://github.com/tinygo-org/tinygo/pull/2375

deadprogram, 2024-05-25:
Closing this, since we cannot support such an old device. Thanks everyone!

closed 2024-05-25 by deadprogram