tinygo-org/tinygo #4212: Can't build for esp32c3, `SCL_PIN` & `SDA_PIN` are undefined

vaaski, 2024-03-28:
Hey, I recently got into Go and loved the idea of using it for microcontrollers in a new project.
I have a bunch of ESP32-C3 superminis but I can't get tinygo to build for them.
I'm on a MacBook Pro M3 Pro on Sonoma with Tinygo and Esptool installed via Homebrew.
I've tried following the blinking LED example adjusted to work with the esp32c3

<details>
<summary>main.go</summary>

```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.GPIO8
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}
```

</details>

But building fails with the following output:
```
❯ tinygo build -target=esp32c3
# machine
/opt/homebrew/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:38:16: undefined: SCL_PIN
/opt/homebrew/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:41:16: undefined: SDA_PIN
```

I guess it's something with the i2c implementation, but I'm not familiar enough with Go or this project to pinpoint the exact issue yet.
Since building with `-target=esp32` works fine, I've tried flashing with that but it won't work either.
```
tinygo flash -target=esp32  
esptool.py v4.7.0
Serial port /dev/cu.usbmodem1101
Connecting...

A fatal error occurred: This chip is ESP32-C3 not ESP32. Wrong --chip argument?
error: failed to flash /var/folders/93/m682qq_n5klc2btq3q_cb_0c0000gq/T/tinygo1642084370/main.bin: exit status 2
```

Thanks a lot for making all this in the first place, I can't wait to use Go for microcontrollers!

vaaski, 2024-04-02:
As a workaround, commenting out these lines makes it work as expected. My LED is blinking.

https://github.com/tinygo-org/tinygo/blob/6384ecace093df2d0b93915886954abfc4ecfe01/src/machine/machine_esp32c3_i2c.go#L37-L42

deadprogram, 2024-04-22:
@vaaski the issue here is that you need to use a target with an actual board file in order to have the correct mapping. An example is the Seeed XIAO ESP32C3 board.

For example, see target https://github.com/tinygo-org/tinygo/blob/release/targets/xiao-esp32c3.json with board file https://github.com/tinygo-org/tinygo/blob/release/src/machine/board_xiao-esp32c3.go

Note that the target "inherits" from the target `esp32c3`. That target is really meant to be used in this way, not as a target of its own.

Hope that helps!

deadprogram, 2024-05-01:
@vaaski if you wanted to create a PR that added the board definitions for the esp32c3 supermini we'd be happy to accept it. :smile: 

See https://forum.arduino.cc/t/esp32-c3-supermini-pinout/1189850/6 for the pin definitions.

See https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board#adding-a-new-board for instructions on how to go about doing this.

deadprogram, 2025-03-16:
Closing since #4518 was merged. Thanks everyone!

PR #4518 (2025-01-20): add board: esp32c3-supermini
Hey, I apologize for the long silence on #4212. I just picked up the project where I need this again and thought I'd give it a shot.

I've used the xiao-esp32c3 and #4454 as a reference.

After building it, I was able to run https://github.com/tinygo-org/tinygo/blob/release/src/examples/blinky1/blinky1.go successfully.

Thanks again for all the work on tinygo and thanks to @deadprogram for the friendly and helpful responses on my original issue. I'm glad I won't have to learn C++ for my project :D 

resolves #4212
closed 2025-03-16 by deadprogram