tinygo-org/tinygo #4216: Flash error on M2 Mac

RiRa12621, 2024-04-03:
## Description
Trying to flash an rp2040. connected via USB-C to MicroUSB cable with the sample code to make the LED blink:

## Code

```golang
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}

```

## Error

```shell
$ tinygo flash -target=nano-rp2040 main.go
error: failed to flash /var/folders/7g/rfh2cs2j4n35ztrch4hfzvq40000gq/T/tinygo658487081/main.uf2: unable to locate any volume: [RPI-RP2]
```

## System

```
$ tinygo version
tinygo version 0.31.2 darwin/arm64 (using go version go1.22.1 and LLVM version 17.0.1)
```

```
$ sudo uname -a
Password:
Darwin F7MQJ63XL2 23.4.0 Darwin Kernel Version 23.4.0: Fri Mar 15 00:12:49 PDT 2024; root:xnu-10063.101.17~1/RELEASE_ARM64_T6020 arm64
```

RiRa12621, 2024-04-03:
Quickly verified that there's no issue with the arduino itself by using the arduino IDE with a blink example:

```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

works as expected

deadprogram, 2024-04-03:
Via Slack:
you probably need to manually kick-in the bootloader the first time flashing that particular board
> Hold down the BOOTSEL button while plugging the board into USB.
you only need to do that once. afterwards, tinygo will detect it is being flashed and kick into bootloader mode
usually if the rp2040 board has already been flashed with something like arduino, it will also work without manual bootloader

RiRa12621, 2024-04-03:
local machine issue: fixed by reboot

closed 2024-04-03 by RiRa12621