tinygo-org/tinygo #4222: tinygo can't compile wasm file

iltatarin, 2024-04-09:
When I want compile my golang app a get errors:
`/go/pkg/mod/golang.org/x/net@v0.24.0/http2/transport.go:26:2: package net/http/httptrace is not in std (/root/.cache/tinygo/goroot-646f794f1c545d78f790d72fe3114fa1e0882f6cfab7deac934ba87ff42a6934/src/net/http/httptrace)
/go/pkg/mod/google.golang.org/grpc@v1.63.2/internal/transport/proxy.go:29:2: package net/http/httputil is not in std (/root/.cache/tinygo/goroot-646f794f1c545d78f790d72fe3114fa1e0882f6cfab7deac934ba87ff42a6934/src/net/http/httputil)`

Golang version: go version go1.21.6 linux/amd64
tinygo version: tinygo version 0.31.2 linux/amd64 (using go version go1.21.6 and LLVM version 17.0.1)
My command: `tinygo build --no-debug -scheduler=none -target=wasi -o filter.wasm cmd/filter.go`
[image]

If I use tinygo version 0.29 I get errors:
[image]

orsinium, 2024-04-11:
WASI preview 1 does not provide enough functions to implement a full HTTP server. More info:

https://go.dev/blog/wasi#limitations

You can use the main Go compiler with [stealthrocket/net](https://github.com/stealthrocket/net) if your target wasm runtime is wasmedge.

brendandburns, 2024-07-08:
Please see this:

https://github.com/dev-wasm/dev-wasm-go?tab=readme-ov-file#wasm-web-serving-with-wasi-http

For a way to use the wasi-http library.

closed 2024-12-02 by deadprogram