tinygo-org/tinygo #4225: tinygo source displeases linters

dkegel-fastly, 2024-04-12:
Eagle-eyed new contributors have noticed that the tinygo source tree does not pass some lint checks.

The project probably shouldn't rely on volunteers to run lint.   

To address this, we might run lint in ci and/or add a "make lint" target in GNUMakefile.

To avoid adding too many new dependencies, we might want to keep the number of linters small, e.g. just go vet and revive, for starters, and avoid metalinters.

dkegel-fastly, 2024-04-26:
Plain old lint target exists, runs one linter now, and several problems have been fixed.

I think we can close this now; further lint enhancements can continue without this bug report being open.

deadprogram, 2024-06-23:
This was released with `v0.32.0` so now closing. Thank you everyone!

closed 2024-06-23 by deadprogram