tinygo-org/tinygo #4250: Undefined os/signal symbols

apprehensions, 2024-04-29:
```
ld.lld: error: undefined symbol: os/signal.signal_recv
>>> referenced by signal_unix.go:23 (/usr/lib/go/src/os/signal/signal_unix.go:23)
>>>               /home/sewn/.cache/tinygo/thinlto/llvmcache-B36A14B3D75DCA16AD65D5B615A6422EA6CEE839:(os/signal.loop)

ld.lld: error: undefined symbol: os/signal.signal_enable
>>> referenced by signal_unix.go:49 (/usr/lib/go/src/os/signal/signal_unix.go:49)
>>>               /home/sewn/.cache/tinygo/thinlto/llvmcache-B36A14B3D75DCA16AD65D5B615A6422EA6CEE839:(os/signal.Notify)

ld.lld: error: undefined symbol: os/signal.signal_disable
>>> referenced by signal_unix.go:53 (/usr/lib/go/src/os/signal/signal_unix.go:53)
>>>               /home/sewn/.cache/tinygo/thinlto/llvmcache-B36A14B3D75DCA16AD65D5B615A6422EA6CEE839:(os/signal.Stop)

ld.lld: error: undefined symbol: os/signal.signalWaitUntilIdle
>>> referenced by signal.go:214 (/usr/lib/go/src/os/signal/signal.go:214)
>>>               /home/sewn/.cache/tinygo/thinlto/llvmcache-B36A14B3D75DCA16AD65D5B615A6422EA6CEE839:(os/signal.Stop)
failed to run tool: ld.lld
error: failed to link /tmp/tinygo3329328202/main: exit status 1
```

4xposed, 2024-07-29:
did you manage to work around this? I'm running into the same issue

aykevl, 2024-08-02:
See #4378 and #4322

deadprogram, 2024-12-06:
Closing since this appears to have been solved. Please reopen if needed. Thanks!

closed 2024-12-06 by deadprogram