tinygo-org/tinygo #4267: hid-keyboard not working on nvidia agx orin

dwSun, 2024-05-20:
I've created a small keyboard using the RP2040, utilizing the code from hid-keyboard. This keyboard has been successfully tested on every other computer in the office, including:

- Linux: Debian Sid, Ubuntu 20.04, Ubuntu 18.04
- Mac: the latest OS version
- Windows: Windows 11
- Embedded device: Jetson Nano

It functions properly on all the aforementioned hardware. However, on the NVIDIA AGX Orin (running Ubuntu 20.04 with JetPack 5.1.2), the keyboard is not recognized, and it doesn't even show up when using lsusb.

Could you please advise where I should start troubleshooting to enable this keyboard to work correctly on the NVIDIA AGX Orin?

sago35, 2024-05-20:
Is the NVIDIA AGX Orin you mentioned the one from the following URL? Does it recognize other keyboards? For example, does it recognize ones made with Arduino?

https://connecttech.com/product/hadron-carrier-for-nvidia-jetson-orin-nx/

If it works with Arduino but not with TinyGo's hid-keyboard, there is likely a difference in USB enumeration. I would like you to try creating a simple keyboard with Arduino to see if it works. @dwSun

dwSun, 2024-05-21:
The device you've linked is one type, whereas I'm using a different model. According to the tests conducted by the device supplier,

NVIDIA Orin devices can recognize standard keyboards and mice, but they are unable to identify the tinygo hid-keyboard.

I don't have an Arduino device at hand; however, I do possess an ESP32-C3 device. When attempting to compile, the following error message is displayed:

```
$ tinygo flash -target=esp32c3 .                                                                                                           [14:30:40]
# machine
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:38:16: undefined: SCL_PIN
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:41:16: undefined: SDA_PIN
```

I will attempt to use an Arduino Nano to see if it succeeds.

deadprogram, 2024-05-21:
> `tinygo flash -target=esp32c3 .`

You need to specify an actual board, you cannot just name the processor. For example:

```shell
tinygo flash -target xiao-esp32c3 .
```

sago35, 2024-05-21:
You can create an Arduino environment using the RP2040. If you let me know which RP2040 board you are using, I might be able to create it as well.

dwSun, 2024-05-21:
> > `tinygo flash -target=esp32c3 .`
> 
> You need to specify an actual board, you cannot just name the processor. For example:
> 
> ```shell
> tinygo flash -target xiao-esp32c3 .
> ```

Thank you, but I have tried all possible device targets and none have been successful.
/usr/local/Cellar/tinygo/0.31.2/targets:

```
(base) [david@Mass]: ~/cljkeyboard-esp32
$ tinygo flash -target=xiao-esp32c3 .                                [18:55:48]
# machine/usb/hid
/usr/local/Cellar/tinygo/0.31.2/src/machine/usb/hid/hid.go:37:11: undefined: machine.ConfigureUSBEndpoint
/usr/local/Cellar/tinygo/0.31.2/src/machine/usb/hid/hid.go:91:11: undefined: machine.SendZlp
$ tinygo flash -target=qtpy-esp32c3 .                                                                                                      [18:59:57]
# machine/usb/hid
/usr/local/Cellar/tinygo/0.31.2/src/machine/usb/hid/hid.go:37:11: undefined: machine.ConfigureUSBEndpoint
/usr/local/Cellar/tinygo/0.31.2/src/machine/usb/hid/hid.go:91:11: undefined: machine.SendZlp
/usr/local/Cellar/tinygo/0.31.2/src/machine/usb/hid/hid.go:99:10: undefined: machine.SendUSBInPacket
(base) [david@Mass]: ~/cljkeyboard-esp32 FAIL
$ tinygo flash -target=makerfabs-esp32c3spi35 .                                                                                            [19:00:07]
# machine
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:38:16: undefined: SCL_PIN
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:41:16: undefined: SDA_PIN
(base) [david@Mass]: ~/cljkeyboard-esp32 FAIL
$ tinygo flash -target=esp32c3-12f .                                                                                                       [19:00:21]
# machine
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:38:16: undefined: SCL_PIN
/usr/local/Cellar/tinygo/0.31.2/src/machine/machine_esp32c3_i2c.go:41:16: undefined: SDA_PIN
(base) [david@Mass]: ~/cljkeyboard-esp32 FAIL
```

dwSun, 2024-05-21:
> You can create an Arduino environment using the RP2040. If you let me know which RP2040 board you are using, I might be able to create it as well.

thanks

I am using waveshare-rp2040-zero to flash the board.
The board I am using come from [this URL](https://item.taobao.com/item.htm?_u=lrupj7ub093&id=766274159050&spm=a1z09.2.0.0.60982e8dvMwZ0Q&skuId=5438794051490), should be compatible with [this](https://www.waveshare.net/wiki/RP2040-Zero)

sago35, 2024-05-21:
The following is not made by me, but it's firmware for a keyboard using the RP2040-Zero. It is created with QMK Vial. Please try writing this UF2 to your board and see if it is recognized as a keyboard.
GP1 to GP14 are COL1 to COL14. GP15 and GP26 to GP29 are ROW1 to ROW5.

https://darakuneko.booth.pm/items/4557796
https://github.com/darakuneko/keyboard/raw/main/firmware/yamada67_an_vial.uf2

dwSun, 2024-05-22:
> The following is not made by me, but it's firmware for a keyboard using the RP2040-Zero. It is created with QMK Vial. Please try writing this UF2 to your board and see if it is recognized as a [...]

thanks.

This firmware is functional, recognizable, and supports character input, having been tested on the following systems:
- Debian Sid
- The latest version of macOS
- NVIDIA AGX Orin (running Ubuntu 20.04 with JetPack 5.1.2)

Within these systems, the keyboard is identified as:
```
Bus 020 Device 049: ID feed:1532 feed Yamada67 AN  Serial Number: vial:f64c2b3c
```

Accordingly, the failure to recognize the `hid-keyboard` from TinyGo suggests that there might be an issue with the code somewhere?

dwSun, 2024-05-22:
I have created a custom keyboard using QMK firmware, and in terms of key functions alone, this keyboard already satisfies my usage requirements. However, implementing custom features like LED blinking and buzzer control with QMK code can be somewhat cumbersome. I remain interested in utilizing TinyGo for this custom keyboard project. Therefore, please continue to attempt to resolve this issue when it is convenient and time permits.

thanks.

sago35, 2024-05-23:
@dwSun
There's one thing I'd like you to try. Does examples/echo2 work on the NVIDIA AGX Orin?

I suspect one of the following causes:

1. The descriptor sent by the microcontroller is different from what the AGX Orin expects.
2. The microcontroller is unable to respond to the commands sent by the AGX Orin.

dwSun, 2024-05-24:
@sago35

To write the firmware, use the following command:
```
tinygo flash -target=waveshare-rp2040-zero -serial usb examples/echo2
```

On a Mac, it is recognized as:
```
Bus 020 Device 029: ID 2e8a:0003 2e8a RP2040-Zero
```

On Debian Sid, it is identified as:
```
Bus 001 Device 009: ID 2e8a:0003 Waveshare RP2040-Zero
```

On Ubuntu 20.04 (x86), it is recognized as:
```
Bus 002 Device 003: ID 2e8a:0003
```

On all the above-tested machines, the device responds to the initial input but fails to respond to subsequent inputs, requiring a re-plug to again react to the first input and then ignore further inputs.

On Ubuntu 18.04 (Jetson Nano), it is identified as:
```
Bus 001 Device 003: ID 2e8a:0003
```
On this device, it only acknowledges with "AT" upon plug-in; manual inputs thereafter receive no response.

On both Ubuntu 20.04 NVIDIA AGX Orin (JetPack 5.1.2) and Ubuntu 20.04 NVIDIA Orin Nano (JetPack 5.1.1), the device is not recognized, and cannot be found via `lsusb`.

sago35, 2024-06-19:
I plan to make the USB enumeration process more similar to what is done with QMK Vial. I will create this in the near future, so please test the connection when it is ready. @dwSun