tinygo-org/tinygo #4269: Does tinygo not support TLS features?

jrovis, 2024-05-21:
Does tinygo not support TLS features?

My Question:
tinygo build -o main.wasm -target=wasi main.go                                                                                         ─╯
# github.com/segmentio/kafka-go/protocol
../../../../go/pkg/mod/github.com/segmentio/kafka-go@v0.4.47/protocol/response.go:156:24: undefined: tls.VersionTLS13
../../../../go/pkg/mod/github.com/segmentio/kafka-go@v0.4.47/protocol/response.go:156:55: undefined: tls.VersionTLS10

Compile environment:
go version go1.22.3 darwin/amd64
tinygo version 0.31.2 darwin/amd64 (using go version go1.22.3 and LLVM version 17.0.1)
kafka-go v0.4.47

labulakalia, 2024-12-28:
https://github.com/tinygo-org/tinygo/blob/release/src/crypto/tls/tls.go#L27
not implemented

scottfeldman, 2025-01-18:
Looking at the TLS usage in kafka-go, I don't believe this is possible with the current "crypto/tls" support in TinyGo.  The TLS support in current TinyGo is limited to baremetal devices with mbed firmware to do the real TLS protocol.  WASM and TLS aren't going to mix with the current state of affairs.  A full port of crypto/tls would fix this.