tinygo-org/tinygo #4275: tinygo build -gc none fails

pierrec, 2024-05-26:
tinygo version 0.31.2 darwin/amd64 (using go version go1.22.3 and LLVM version 17.0.1)

```
tinygo build -gc none
ld.lld: error: undefined symbol: _runtime.alloc
>>> referenced by task_stack.go:75 (/usr/local/Cellar/tinygo/0.31.2/src/internal/task/task_stack.go:75)
>>>               /tmp/lto.tmp:(symbol _internal/task.start+0x19)
>>> referenced by task_stack.go:113 (/usr/local/Cellar/tinygo/0.31.2/src/internal/task/task_stack.go:113)
>>>               /tmp/lto.tmp:(symbol _internal/task.start+0xa)
failed to run tool: ld.lld
error: failed to link /var/folders/lb/nhsppbxs5_v27pn_x3kf66wc0000gn/T/tinygo1324987751/main: exit status 1
```

Note it works fine if used in combination of `-scheduler none`

dgryski, 2024-05-31:
If you need a minimal garbage collector try `-gc=leaking`.

davecheney, 2026-10-27:
Thanks for reporting this. I retested with the current TinyGo tree using:

```go
package main

func main() {}
```

The original `tinygo build -gc=none main.go` command now succeeds, including when targeting macOS/amd64. The default scheduler is now `threads`.

Explicitly selecting `-scheduler=tasks` still produces linker errors. I’ve opened #6051 to track providing a clearer diagnostic for that combination.

Closing this as resolved for the original command. Please feel free to reopen if your program still encounters this failure.

closed 2026-10-27 by davecheney