tinygo-org/tinygo #4300: TypeError: WebAssembly.instantiate(): Import #0 "gojs": module is not an object or function  TypeError: Cannot convert a BigInt value to a number

zzxap, 2024-06-17:
TypeError: WebAssembly.instantiate(): Import #0 "gojs": module is not an object or function

if  change to

"syscall/js.finalizeRef": (v_addr) => {
						// Note: TinyGo does not support finalizers so this is only called
						// for one specific case, by js.go:jsString.
						// https://github.com/tinygo-org/tinygo/issues/1140
						const id = mem().getUint32(v_addr, true);
						this._goRefCounts[id]--;
						if (this._goRefCounts[id] === 0) {
							const v = this._values[id];
							this._values[id] = null;
							this._ids.delete(v);
							this._idPool.push(id);
						}
					},
will got
TypeError: Cannot convert a BigInt value to a number
    at DataView.prototype.getUint32 (<anonymous>)

tinygo version 0.31.2 darwin/amd64 (using go version go1.22.3 and LLVM version 17.0.1)

aykevl, 2024-06-21:
It looks like you are using an old version of wasm_exec.js. Please use the version that is part of the TinyGo version that you use to build the WebAssembly file.

deadprogram, 2024-06-27:
Now closing as answered. Thanks everyone!

closed 2024-06-27 by deadprogram