tinygo-org/tinygo #4314: build with target "wasm-unknown" fail in windows10 22H2

longkui-clown, 2024-06-27:
when I use cmd `tinygo build -o demo_go.wasm -target wasm-unknown -no-debug main.go` to build wasm file, it return errors like below:
```
# os
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\types_unix.go:20:18: undefined: syscall.Stat_t
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:50:32: undefined: syscall.Timespec
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dir_unix.go:55:28: undefined: syscall.ReadDirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dir_unix.go:82:49: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:15:46: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:15:83: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:19:46: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:19:86: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:27:49: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:31:33: undefined: syscall.Dirent
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:37:15: undefined: syscall.DT_BLK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:39:15: undefined: syscall.DT_CHR
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:41:15: undefined: syscall.DT_DIR
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:43:15: undefined: syscall.DT_FIFO
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:45:15: undefined: syscall.DT_LNK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:47:15: undefined: syscall.DT_REG
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\dirent_linux.go:49:15: undefined: syscall.DT_SOCK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:35:18: undefined: syscall.Chdir
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:55:36: undefined: syscall.Mkdir
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:63:34: undefined: syscall.Unlink
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:67:35: undefined: syscall.Rmdir
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:110:19: undefined: syscall.Write
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:142:18: undefined: syscall.Chmod
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:186:16: undefined: syscall.S_ISUID
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:189:16: undefined: syscall.S_ISGID
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_anyos.go:192:16: undefined: syscall.S_ISVTX
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:28:17: undefined: syscall.Rename
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:81:18: undefined: syscall.Link
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:96:18: undefined: syscall.Symlink
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:114:28: undefined: syscall.Readlink
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:114:45: not enough arguments in call to fixCount
        have (unknown type)
        want (int, error)
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:133:19: undefined: syscall.Pread
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:149:20: undefined: syscall.Pwrite
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\file_unix.go:160:17: undefined: syscall.Fsync
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:23:31: undefined: syscall.S_IFMT
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:24:15: undefined: syscall.S_IFBLK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:26:15: undefined: syscall.S_IFCHR
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:28:15: undefined: syscall.S_IFDIR
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:30:15: undefined: syscall.S_IFIFO
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:32:15: undefined: syscall.S_IFLNK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:34:15: undefined: syscall.S_IFREG
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:36:15: undefined: syscall.S_IFSOCK
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:39:25: undefined: syscall.S_ISGID
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:42:25: undefined: syscall.S_ISUID
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:45:25: undefined: syscall.S_ISVTX
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_linuxlike.go:56:43: undefined: syscall.Stat_t
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_unix.go:18:18: undefined: syscall.Fstat
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_unix.go:31:37: undefined: syscall.Stat
C:\tinygo0.32.0.windows-amd64\tinygo\src\os\stat_unix.go:44:37: undefined: syscall.Lstat
```

environment:
```
go version
go version go1.20.2 windows/amd64

tinygo.exe version
tinygo version 0.32.0 windows/amd64 (using go version go1.20.2 and LLVM version 18.1.2)

binaryen-version
117
```

aykevl, 2024-06-27:
See #4315 where I've made a PR to fix this issue.

longkui-clown, 2024-06-28:
thanks, it works @aykevl

deadprogram, 2024-12-06:
Closing this this was addressed. Thanks!

closed 2024-12-06 by deadprogram