tinygo-org/tinygo #4319: wasm-unknown target shouldn't set `--import-memory` by default

jmillikin, 2024-06-30:
The `wasm-unknown` target invokes `wasm-ld` with the [`--import-memory`](https://lld.llvm.org/WebAssembly.html#cmdoption-import-memory) option, which causes the following import to be emitted in the output:

```wat
(import "env" "memory" (memory $mimport$0 2))
```

This is undesirable to have as default behavior for `wasm-unknown`, because it introduces a mandatory import into what is otherwise a freestanding module.

If the `--import-memory` were to be removed from `targets/wasm-unknown.json` ldflags, then `wasm-ld` would emit this instead:

```wat
;; Declare and export the Memory, which requires no special import binding in the host
(memory $0 2)
(export "memory" (memory $0))
```

This option appears to have been added as part of a larger change in https://github.com/tinygo-org/tinygo/pull/4125/commits/2cd966d6d577df17cfde3869ee7e0dbb4da9307f, but it's unclear from the commit message why `--import-memory` was added. Based on discussion in the PR, I think it was simply to reduce the number of bytes in an absolute bare-bones "hello world" module, which IMO is not a compelling reason to have it set by default.

aykevl, 2024-06-30:
Because wasm-unknown by default doesn't target a particular platform, it's difficult to say whether memory should be imported or exported by default.
@deadprogram what do environments like wazero expect? Memory, or no memory? Or do they support both?

deadprogram, 2024-07-17:
I think @jmillikin is probably correct in what the default should be.

deadprogram, 2024-08-16:
See https://github.com/tinygo-org/tinygo/issues/4319 for a correction.

commit 16950780 (2024-08-17): targets: remove import-memory flag from wasm-unknown target to fix #4319
closed 2024-08-20 by deadprogram