tinygo-org/tinygo #4320: Ability to pass flags to external linker (such as `wasm-ld`)

jmillikin, 2024-06-30:
While looking into https://github.com/tinygo-org/tinygo/issues/4319, I discovered that there doesn't seem to be a way to pass flags to the `wasm-ld` external linker from the `tinygo build` command. The only way I could find is to create a custom target JSON, which is a bit heavyweight.

There is `-ldflags`, but it's documented as being for flags compatible with the Go linker.

The standard Go build system defines `-extldflags` which seems to be what I want, but `tinygo build` doesn't have that flag.

dgryski, 2024-07-03:
I can probably take a look at this.

aykevl, 2024-10-22:
There is also `#cgo LDFLAGS` which may be appropriate in some circumstances.

PR #4472 (2024-09-18): support -extldflags
Fixes #4320
commit 892efaec (2024-09-18): support -extldflags

Fixes #4320
closed 2024-10-25 by aykevl