tinygo-org/tinygo #4328: cannot use `tinygo build -ldflags "-X 'main.WASM_NAME=ws_conn.wasm'"` to modify variable WASM_NAME

longkui-clown, 2024-07-08:
when i use cmd `tinygo build -ldflags "-X 'main.WASM_NAME=ws_conn.wasm'"` to change the variable `WASM_NAME` in main package like golang, i got the return val in default val, is it not work?
[image]
[image]

Tush-r, 2024-08-26:
The documentation mentions that if a default value is set for a variable, value passed with ldflags will be ignored. 

```
One important thing to note is that you cannot use a default value in your code if you want to be able to set that value using -ldflags. If you set the variable to a default value, then the value you pass using ldflags will be ignored.
```
Source: [How to set build-time variables](https://tinygo.org/docs/guides/tips-n-tricks/)

longkui-clown, 2024-08-27:
thanks，i got it

deadprogram, 2024-09-06:
Closing since the question was answered. Thank you!

closed 2024-09-06 by deadprogram