tinygo-org/tinygo #4341: "Over The Air" updates for Pico-w

nickaxgit, 2024-07-16:
Hi Chaps

With some direction from Patricio, I have implemented the UDP stack in seqs.. I can now pull binary files with a naïve, "reliable" protocol on top. (Yes, I know I could have done that with TCP !). 
I made it for camera images, but the next challenge is to be able to update the running program - I'm not sure I need anything as complex as a bootloader - if I can get a new .ELF (or UF2) file onto the Pico, and invoke machine.CPUReset() ... ??
I also just discovered machine.BlockDevice 

I'm sure this is not as straightforward as I am hoping, but I'd welcome any pointers.

Is self-updating for TinyGo apps already elsewhere on the radar ?
My needs are simple, it doesn't need to be totally robust or secure, bricking a Pico would just mean a short drive - but I could perhaps explore a Proof of Concept that might develop into something more universally useful.

nickaxgit, 2024-07-16:
Note to self (just gathering potentially helpful stuff)
https://kevinboone.me/pico_run_ram2.html

b0ch3nski, 2024-07-17:
ESPhome has a support for both OTA updates and RP2040 - you could take a look how they are doing it:
1. https://esphome.io/components/ota
2. https://esphome.io/components/safe_mode
3. https://esphome.io/components/rp2040

IMO for a reliable solution, this would require some sort of bootloader and A/B partitions switchover :thinking:

nickaxgit, 2024-07-17:
Thanks… will check it.
There are a few C solutions out there, some look very robust.. I was
looking at something much simpler with a smaller footprint, thinking there
might be a way to live very dangerously and overwrite the program that is
executing in a single memcopy, rebooting with a watchdog timer.

Perhaps it’s better to forget such a terrible idea and put the effort into
something more robust and resilient, it would be nice if it was all GO
though.

deadprogram, 2024-12-02:
Closing as question was answered. Thanks everyone!