tinygo-org/tinygo #4346: bug: net/http server not working on linux

ldemailly, 2024-07-18:
```go
package main

import (
    "net/http"
    "log"
)

func helloHandler(w http.ResponseWriter, r *http.Request) {
    w.Write([]byte("Hello, TinyGo!"))
}

func main() {
    mux := http.NewServeMux()
    mux.HandleFunc("/", helloHandler)
    log.Fatal(http.ListenAndServe(":8080", mux))
}
```

```shell
tinygo run .
```

```
panic: runtime error at 0x0000000000233e7f: nil pointer dereference
error: failed to run compiled binary /tmp/tinygo1722875622/main: signal: aborted (core dumped)
```

dkegel-fastly, 2024-07-19:
This is, alas, documented at https://tinygo.org/docs/reference/lang-support/stdlib/#nethttp

On the bright side, the list of dependencies that can't be imported is now down to two; it was six three years ago, see https://web.archive.org/web/20210616011030/https://tinygo.org/docs/reference/lang-support/stdlib/#nethttp

ldemailly, 2024-07-19:
well there should be a way to isolate the crypto part no?

http is pretty basic

dkegel-fastly, 2024-07-19:
Yeah, that page needs some updating.    If the problem is still crypto related, as I think dgryski mentioned, a stub of the offending crypto module might at least let plain old http work.

davecheney, 2026-10-28:
This now works with TinyGo 0.44-dev. I cross-compiled the following program for Linux/AMD64 and ran it on a Linux host:

```go
package main

import (
	"log"
	"net/http"
)

func main() {
	mux := http.NewServeMux()
	mux.HandleFunc("/", func(w http.ResponseWriter, r *http.Request) {
		w.Write([]byte("Hello, TinyGo!"))
	})
	log.Fatal(http.ListenAndServe(":8080", mux))
}
```

```sh
GOOS=linux GOARCH=amd64 tinygo build -o server main.go
./server
```

A request to `http://127.0.0.1:8080/` returned:

```text
HTTP 200
Hello, TinyGo!
```

Closing as fixed. Please reopen if you still encounter this with a current version of TinyGo.

closed 2026-10-28 by davecheney