tinygo-org/tinygo #4347: ci:  packages have unmet dependecies

leongross, 2024-07-19:
While the [ci](https://github.com/tinygo-org/tinygo/actions/runs/10005066422/job/27655004450?pr=4213) for my PR run, I realized that the dependency setup failed.

```
Reading package lists...
W: https://apt.llvm.org/jammy/dists/llvm-toolchain-jammy-18/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libllvm18 : Breaks: llvm-18-dev (< 1:18.1.8-4) but 1:18.1.8~++20240717051017+3b5b5c1ec4a3-1~exp1~20240717171122.141 is to be installed
E: Unable to correct problems, you have held broken packages.
Error: Process completed with exit code 100.
```

I will have a look at that soon.

leongross, 2024-07-19:
Update: There seems to be a general problem regarding the llvm packages provided via apt

https://github.com/llvm/llvm-project/issues/99453

deadprogram, 2024-07-19:
https://github.com/llvm/llvm-project/issues/99502 despite being a duplicate actually has more info. Supposed to be fixed now, but rerunning the tinygo ci check still shows same error.

leongross, 2024-07-19:
Does the CI use the updated package index of `llvm` or is this maybe some caching issue?

deadprogram, 2024-07-20:
This has now been corrected by #4350 so now closing. Thank you for reporting @leongross

closed 2024-07-20 by deadprogram