tinygo-org/tinygo #4353: runtime.hashmapStringGet segv

ldemailly, 2024-07-21:
```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2ffffffe0)
    frame #0: 0x0000000100008030 tiny_test`runtime.hashmapStringGet
tiny_test`runtime.hashmapStringGet:
->  0x100008030 <+0>:  stp    x24, x23, [sp, #-0x40]!
    0x100008034 <+4>:  stp    x22, x21, [sp, #0x10]
    0x100008038 <+8>:  stp    x20, x19, [sp, #0x20]
    0x10000803c <+12>: stp    x29, x30, [sp, #0x30]
Target 0: (tiny_test) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2ffffffe0)
  * frame #0: 0x0000000100008030 tiny_test`runtime.hashmapStringGet
    frame #1: 0x000000010002b218 tiny_test`(*grol.io/grol/object.Environment).Get + 48
    frame #2: 0x000000010002b53c tiny_test`(*grol.io/grol/eval.State).evalInternal + 628
    frame #3: 0x000000010002b2ac tiny_test`(*grol.io/grol/eval.State).Eval + 8
    frame #4: 0x000000010002b754 tiny_test`(*grol.io/grol/eval.State).evalInternal + 1164
    frame #5: 0x000000010002b4c8 tiny_test`(*grol.io/grol/eval.State).evalInternal + 512
```

to repro:
```
tinygo test -x -v --tags no_net,no_json -c -o tiny_test grol.io/grol/bug
```
(will edit with link shortly)

ldemailly, 2024-07-21:
Thanks Damian:

```
tinygo build -stack-size=10MB -o grol.tiny -tags no_net,no_json .
gdb grol.tiny 
(gdb) run large_fact.gr 
Starting program: /home/moorea/github/grol/grol.tiny large_fact.gr
17:48:26.784 grol dev  tinygo0.33.0-dev-f14339de amd64 linux - welcome!
17:48:26.784 [INF] Running large_fact.gr
30414093201713376000000000000000000000000000000000000000000000000
17:48:26.785 [INF] All done
[Inferior 1 (process 124933) exited normally]
(gdb) 
```
works...

what is the default stack value? on x86/macOS/wasm?

ldemailly, 2024-07-22:
default is 64k yet somehow on wasm it's a lot less, or a chunk of it is taken before it gets to user code

this is sort of a dup of #3258

aykevl, 2024-07-22:
> default is 64k yet somehow on wasm it's a lot less, or a chunk of it is taken before it gets to user code

WebAssembly currently requires a lot more stack space because most local pointer values are stored on the stack (so they can be scanned by the GC). If you use `-gc=leaking` you'll probably see you have a lot more stack space.

ldemailly, 2024-08-14:
Closing as this is sort of solved by passing stack size argument

closed 2024-08-14 by ldemailly