tinygo-org/tinygo #4357: Flash USB from https://play.tinygo.org/ on Chromebook?

GeoffreyPlitt, 2024-07-22:
Hi! I'm trying to flash a Pi Pico attached via USB, from https://play.tinygo.org/, on Chromebook, which supports WebUSB . The button doesn't seem to do anything. Is it supposed to work?

GeoffreyPlitt, 2024-07-23:
How does the "flash" button work then?

deadprogram, 2024-07-23:
It should compile and download the UF2 file to your normal "Downloads" directory.

deadprogram, 2024-12-07:
Closing since this question was answered. Thanks!

closed 2024-12-07 by deadprogram

aykevl, 2024-12-14:
The idea is that you put the device in firmware download mode (often by double pressing the reset button, though for the Pico you need to keep the "bootsel" button pressed while connecting via USB). Then it will present itself as a mass storage device. You can then download the file to this device like any other file you download from the internet.