tinygo-org/tinygo #4396: Add `node:` Prefix to Node.js API Imports in `wasm_exec.js` for Cloudflare Workers Compatibility

monta-k, 2024-08-08:
When using TinyGo with Cloudflare Workers and the compatibility_flags = ["nodejs_compat"] setting, it is necessary to import Node.js APIs with the node: prefix. The current wasm_exec.js file provided by TinyGo does not include this prefix, causing compatibility issues.
Update the wasm_exec.js file to include the node: prefix for all Node.js API imports. This change will ensure compatibility with Cloudflare Workers and adhere to the required import syntax.

Example
Current:
```
global.fs = require("fs");
```
Updated import:
```
global.fs = require("node:fs");
```

Would it be okay for me to create a pull request for this change?
Thank you for considering this enhancement.

monta-k, 2024-08-08:
Here is the `wasm_exec.js` file that has been confirmed to work with Cloudflare Workers with the `compatibility_flags = ["nodejs_compat"]` setting.
https://github.com/monta-k/cloudflare_workers_tinygo_wasm_exec

aykevl, 2024-08-08:
I looked into it a bit and this seems to be a standard way to get them. Feel free to make a PR with this change. If it works with Node.js, I don't see why it would be a problem.

closed 2024-08-09 by monta-k