tinygo-org/tinygo #4416: add support for GODEBUG=gotypesalias=1

eliasnaur, 2024-08-17:
Seeing https://github.com/tinygo-org/tinygo/pull/4305 merged, I tested tinygo with Go 1.23:

```sh
$ git diff
diff --git a/go.mod b/go.mod
index bf85ef3a..a484fcf2 100644
--- a/go.mod
+++ b/go.mod
@@ -1,6 +1,8 @@
 module github.com/tinygo-org/tinygo

-go 1.19
+go 1.23
+
+toolchain go1.23.0

 require (
        github.com/aykevl/go-wasm v0.0.2-0.20240312204833-50275154210c
diff --git a/testdata/go1.23/go.mod b/testdata/go1.23/go.mod
index c0ad79b6..e2426544 100644
--- a/testdata/go1.23/go.mod
+++ b/testdata/go1.23/go.mod
@@ -1,3 +1,5 @@
 module github.com/tinygo-org/tinygo/testdata/go1.23

 go 1.23
+
+toolchain go1.23.0
$ go install
$ cd testdata/go1.23
$ tinygo build
panic: unknown type: any

goroutine 102 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0x140010a0280?, {0x102f83488?, 0x140001de080?})
	/Users/a/proj/tinygo/compiler/compiler.go:466 +0x590
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType(0x140010a0280, {0x102f83488, 0x140001de080})
	/Users/a/proj/tinygo/compiler/compiler.go:381 +0x6c
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0x140010a0280, {0x102f833c0?, 0x14000602e70})
	/Users/a/proj/tinygo/compiler/compiler.go:454 +0x5d4
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType(0x140010a0280, {0x102f833c0, 0x14000602e70})
	/Users/a/proj/tinygo/compiler/compiler.go:381 +0x6c
github.com/tinygo-org/tinygo/compiler.(*compilerContext).makeLLVMType(0x140010a0280, {0x102f83140, 0x140001d7810})
	/Users/a/proj/tinygo/compiler/compiler.go:437 +0x504
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getLLVMType(0x140010a0280, {0x102f83140, 0x140001d7810})
	/Users/a/proj/tinygo/compiler/compiler.go:381 +0x6c
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getParamInfo(0x140010a0280, {0x102d5e0c0?}, {0x103222a70, 0x1}, {0x102f833e8?, 0x140003b32a0?})
	/Users/a/proj/tinygo/compiler/calls.go:203 +0x120
github.com/tinygo-org/tinygo/compiler.(*compilerContext).expandFormalParamType(0x140010a0280, {0x102f833e8?}, {0x103222a70, 0x1}, {0x102f833e8, 0x140003b32a0})
	/Users/a/proj/tinygo/compiler/calls.go:103 +0x84
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0x140010a0280, 0x14000bbb800)
	/Users/a/proj/tinygo/compiler/symbol.go:92 +0x33c
github.com/tinygo-org/tinygo/compiler.newBuilder(0x140010a0280, {0x14000605290?}, 0x14000bbb800)
	/Users/a/proj/tinygo/compiler/compiler.go:177 +0x34
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0x140010a0280, {0x14000c8da00?}, 0x14000894800)
	/Users/a/proj/tinygo/compiler/compiler.go:888 +0x79c
github.com/tinygo-org/tinygo/compiler.CompilePackage({0x14000954070?, 0x5e?}, 0x1400094ef00, 0x14000894800, {0xd8a403833eff6dcd?}, 0x102e23ef3?, 0x68?)
	/Users/a/proj/tinygo/compiler/compiler.go:312 +0x394
github.com/tinygo-org/tinygo/builder.Build.func3(0x1400072ad20)
	/Users/a/proj/tinygo/builder/build.go:375 +0x18c
github.com/tinygo-org/tinygo/builder.runJob(0x1400072ad20, 0x14000950770)
	/Users/a/proj/tinygo/builder/jobs.go:222 +0x48
created by github.com/tinygo-org/tinygo/builder.runJobs in goroutine 1
	/Users/a/proj/tinygo/builder/jobs.go:123 +0x43c
```

The workaround is building with `GODEBUG=gotypesalias=0`, the default in Go 1.22:

```sh
$ GODEBUG=gotypesalias=0 tinygo build
```

Apologies if this is a known issue.

deadprogram, 2024-08-18:
@eliasnaur can you provide a reproducer? thank you.

eliasnaur, 2024-08-18:
Sorry for not being clear: the steps I outlined is from a checkout of the tinygo repository. The test program is your testdata/go1.23.

aykevl, 2025-05-24:
Fix: https://github.com/tinygo-org/tinygo/pull/4911