tinygo-org/tinygo #4421: add support for `-C` flag to chdir before running

archie2x, 2024-08-20:
Supported by big Go since 1.20 for all commands. It's helpful for scripting.  Expected behavior:

- cd's to argument
- runs rest of command

    tinygo run -C path/to/dir some_file.go   # some_file.go is in path/to/dir

See `go help build`:

    % go help build
    usage: go build [-o output] [build flags] [packages]

    [...]

      -C dir
        Change to dir before running the command.
        Any files named on the command line are interpreted after
        changing directories.
        If used, this flag must be the first one in the command line.

    [...]

archie2x, 2024-09-13:
Fixed by 5abf1e9. Can I close?

dgryski, 2024-09-13:
We generally close after the release that includes this fix is out.  I'll tag this issue as "next-release" so it gets handled correctly.

commit 5abf1e99 (2024-09-13): Fix #4421: Add `-C DIR` flag (#4422)

feature: Fix #4421: Add `-C DIR` flag
closed 2024-10-25 by aykevl