tinygo-org/tinygo #4427: Fedora package out of date

AverseABFun, 2024-08-21:
As of right now, [the fedora package](https://packages.fedoraproject.org/pkgs/tinygo/tinygo/) is out of date(it is version 0.30.0, whereas upstream is currently 0.33.0). This also introduces some issues when installing it and go from there as 0.30 only supports go 1.21 and the go version in fedora is 1.22.5 for fedora 40 and 1.23 for rawhide.

deadprogram, 2024-08-22:
@AverseABFun I think that @QuLogic is the package maintainer for Fedora?

QuLogic, 2024-10-28:
Apologies, there were some dependency issues, and legal things to sort out. I pinged them, and will try to get things up-to-date again.

dnlwgnd, 2025-02-24:
ran into this today. guess I manually install a release then.

F0rgenet, 2025-03-29:
Fix please, annoying

deadprogram, 2025-03-30:
Sorry, you need to ask the Fedora package maintainer. We here are the developers.

Maybe @QuLogic can shed some light on this?

GageHoweTamu, 2025-05-14:
Still an issue as of May 14, 2025.

QuLogic, 2025-07-19:
Good news! Legal has finally made a decision on things, so I can start updating this again.

QuLogic, 2025-08-03:
I've updated this in [Rawhide](https://bodhi.fedoraproject.org/updates/FEDORA-2025-a67a1a0035) and working on updating the other branches.

davecheney, 2026-10-27:
Thanks for updating Rawhide. Fedora package updates are handled downstream by the Fedora maintainers. For the status of other branches, please contact tinygo-maintainers@fedoraproject.org. Closing this upstream issue since there’s no TinyGo source change to track.

closed 2026-10-27 by davecheney