tinygo-org/tinygo #4434: Adding Arena Allocator?

s3rj1k, 2024-08-24:
Adding a first-class support for memory arenas could possibly benefit in avoiding some of GC issues especially present in WASM.

aykevl, 2024-08-24:
I don't see how an arena allocator can possibly be used safely in Go. Go is a garbage collected language, so if any outside objects have a pointer to an object inside the arena that will become a dangling pointer after freeing the arena.

I can't prevent anybody from shooting themselves in the foot, but I don't think this is something we're going to add to TinyGo to be honest.

A better solution in my opinion would be to improve the GC itself. For example, I've been working on adding the (very well tuned) Boehm GC to TinyGo: https://github.com/tinygo-org/tinygo/pull/4385
It will take some work to also get it working correctly on WebAssembly (WebAssembly is a difficult system), but I don't see any fundamental blockers why it couldn't work.

s3rj1k, 2024-08-24:
@aykevl What about support of implicit `free` for some allocation, so that most of the work is GC based but for some specific cases manual memory management?

aykevl, 2024-08-26:
Can you describe your use case?
Is it perhaps large areas of pointer-free memory that are slow?

closed 2024-08-26 by aykevl

s3rj1k, 2024-08-26:
> Can you describe your use case?

Mainly WASM binaries for web server custom filters. (ref: https://github.com/tetratelabs/proxy-wasm-go-sdk)

aykevl, 2024-08-27:
Hmm, okay. I have seen that comment, but I haven't experienced those issues myself. GC is a known issue in WebAssembly because WebAssembly doesn't allow scanning the stack so it's always going to be less efficient. I hope to make it at least a bit more efficient with bdwgc. Or, _maybe_, it will be possible in the future to use the built-in WebAssembly GC (which is currently entirely unsupported in LLVM so right now is impossible to implement).