tinygo-org/tinygo #4441: compiler crash on defer inside range-over-func

ALTree, 2024-08-28:
```
$ tinygo version
tinygo version 0.33.0 linux/amd64 (using go version go1.23.0 and LLVM version 18.1.2)
```

```
package main

func main() {
	for range func(func() bool) {} {
		defer func() {}()
	}
}
```

```
$ tinygo build crash.go
panic: *ssa.opaqueType: deferStack

goroutine 10 [running]:
golang.org/x/tools/go/types/typeutil.Hasher.hashFor({0xc000585650?, 0xc000585680?, 0x0?}, {0x6524780?, 0x866af40?})
	/go/pkg/mod/golang.org/x/tools@v0.22.1-0.20240621165957-db513b091504/go/types/typeutil/map.go:375 +0x58c
golang.org/x/tools/go/types/typeutil.Hasher.Hash({0xc000585650?, 0xc000585680?, 0x0?}, {0x6524780, 0x866af40})
	/go/pkg/mod/golang.org/x/tools@v0.22.1-0.20240621165957-db513b091504/go/types/typeutil/map.go:240 +0x75
golang.org/x/tools/go/types/typeutil.Hasher.hashFor({0xc000585650?, 0xc000585680?, 0x0?}, {0x6523da8?, 0x866af30})
	/go/pkg/mod/golang.org/x/tools@v0.22.1-0.20240621165957-db513b091504/go/types/typeutil/map.go:286 +0x165
golang.org/x/tools/go/types/typeutil.Hasher.Hash({0xc000585650?, 0xc000585680?, 0x0?}, {0x6523da8, 0x866af30})

[...]
```

deadprogram, 2025-03-16:
Is this issue still value for the most recent TinyGo version?

ALTree, 2025-03-17:
> Is this issue still value for the most recent TinyGo version?

Crashes with the same error on 0.36

aykevl, 2025-03-18:
Reproduced locally.
Interestingly this crash happens while it is trying to report the following error:

```
todo: builtin: ssa:deferstack
```

aykevl, 2025-03-18:
So there's a `ssa:deferstack` call and a `*deferStack` type, and I'm not entirely sure what they do. I will look into it later.
I filed https://github.com/golang/go/issues/72914 related to this issue (fixing that one would make it easier to fix this bug).

davecheney, 2026-10-28:
This is blocked on #5984 due to the chain of dependency bumps