tinygo-org/tinygo #4443: Arduino uno r4

dndeus, 2024-08-28:
Does this work with arduino uno r4??

I'm keep getting 

```
Error: programmer is not responding
Warning: attempt 1 of 10: not in sync: resp=0x00
```

dndeus, 2024-09-08:
Thanks for the update, I noticed you made a commit!!

I guess it's WIP?! it's not merge to main yet I assume

Thank you for your efforts!

deadprogram, 2024-09-09:
I just updated the branch but there is still quite a bit of work needed to make it functional.

dndeus, 2024-09-09:
cool,  thanks for the update!

dgryski, 2026-10-24:
Basic Arduino Uno R4 is now merged https://github.com/tinygo-org/tinygo/pull/5734 .

davecheney, 2026-10-24:
Basic Arduino Uno R4 support has merged in https://github.com/tinygo-org/tinygo/pull/5734. Please raise additional issues for any missing features in the arduino-uno-r4-wifi and arduino-uno-r4-minima targets.

closed 2026-10-24 by davecheney