tinygo-org/tinygo #4446: Does tinygo itself run with WASI?

HeavySnowJakarta, 2024-08-29:
I'm trying for bringing Go support to [a-Shell](https://github.com/holzschu/a-shell). There is a WASI runtime inside the iOS app so if we can compile `tinygo` to WASI and let it generate WASI executables it'll just be fine.

+ There are a dozens of API with WASI on a-Shell to operate files
+ Threads are not available

aykevl, 2024-08-29:
I wish. LLVM is [working on it](https://discourse.llvm.org/t/rfc-building-llvm-for-webassembly/79073/36). But CGo is also needed, and it looks like Go doesn't support CGo yet for WebAssembly (TinyGo does, though!).
So maybe in the future, but not right now.

closed 2024-08-29 by aykevl

HeavySnowJakarta, 2024-08-29:
But tinygo itself generous llvm ir right? or must do with gc?

aykevl, 2024-08-29:
I don't understand what you're saying?
The `tinygo` binary uses LLVM, but is itself built using gc.

HeavySnowJakarta, 2024-08-29:
> I don't understand what you're saying?
> The `tinygo` binary uses LLVM, but is itself built using gc.

errors above.

if i'm not wrong, when tinygo works it generates llvm ir, and then call llvm to generate final binaries, right? i thought tinygo (`guest=wasi`) may work with another custom llvm (`guest=native binary`, `target=wasi`)

aykevl, 2024-08-30:
If I understand you correctly, no it doesn't work that way.

TinyGo does generate IR, but it needs LLVM to do that. It is linked to LLVM. You cannot build TinyGo without also linking to LLVM.

HeavySnowJakarta, 2024-09-05:
> If I understand you correctly, no it doesn't work that way.
> 
> TinyGo does generate IR, but it needs LLVM to do that. It is linked to LLVM. You cannot build TinyGo without also linking to LLVM.

got clarified and thx for your patience. btw how could i know how it's got linked with llvm (during the build)?