tinygo-org/tinygo #4450: Can't find modules in std lib but GOROOT is set.

Joematpal, 2024-08-31:
I set my GOROOT from the tinygo info.

I try to build. The build process looks for modules in the wrong cache directory.

I'm getting these errors. I think
`package sync/atomic is not in std`
`package net/netip is not in std`

```shell
$ joematpal/nano_33_test » tinygo info
LLVM triple:       x86_64-unknown-linux
GOOS:              linux
GOARCH:            amd64
build tags:        linux amd64 tinygo math_big_pure_go gc.precise scheduler.tasks serial.none
garbage collector: precise
scheduler:         tasks
cached GOROOT:     /home/joseph/.cache/tinygo/goroot-a107edff9f5d5423085f0d0157bfbc7b3dc6033d6b19de196de3c0bcd288f8ae

$ joematpal/nano_33_test » go version
go version go1.21.13 linux/amd64
$ joematpal/nano_33_test » go env GOROOT
/home/joseph/.cache/tinygo/goroot-501d65f5c2ecab33a8292a987dfa4379c665c0c2389f5b49c9e4d20d92002090
$ joematpal/nano_33_test » echo $GOROOT
/home/joseph/.cache/tinygo/goroot-501d65f5c2ecab33a8292a987dfa4379c665c0c2389f5b49c9e4d20d92002090

$ joematpal/nano_33_test » tinygo flash -target=arduino-nano33 -port=/dev/ttyACM0
/home/joseph/.cache/tinygo/goroot-7d34bc3707171ae3faefa9f5bc728c78f3f45ce029af31f633ce4498a3ad4394/src/runtime/cond.go:7:2: package sync/atomic is not in std (/home/joseph/.cache/tinygo/goroot-7d34bc3707171ae3faefa9f5bc728c78f3f45ce029af31f633ce4498a3ad4394/src/sync/atomic)
/home/joseph/.cache/tinygo/goroot-7d34bc3707171ae3faefa9f5bc728c78f3f45ce029af31f633ce4498a3ad4394/src/net/tcpsock.go:5:2: package net/netip is not in std (/home/joseph/.cache/tinygo/goroot-7d34bc3707171ae3faefa9f5bc728c78f3f45ce029af31f633ce4498a3ad4394/src/net/netip)
```

I was using the latest version of tinygo and the same thing was happening so moved to 0.29

I'm sure I'm missing something in my setup. If anyone could help that would be great.

Why do new tinygo caches keep getting added to the cache dir? What is the rational around that? I just cleared everything and it just keeps growing. I feel like everytime I run `tinygo flash` it adds a new one.

Joematpal, 2024-08-31:
I saw that the tinygo GOROOT in the info command had just changed. So I re ran it.

```shell
$ joematpal/nano_33_test » tinygo flash -target=arduino-nano33 -port=/dev/ttyACM0                    
error: open /home/joseph/.cache/tinygo/goroot-a107edff9f5d5423085f0d0157bfbc7b3dc6033d6b19de196de3c0bcd288f8ae/src/crypto/internal/boring: no such file or directory
```
I got a new error. I haven't seen this.

Joematpal, 2024-08-31:
I removed all of the GOROOTs that i had all over the place. I see now the differences in the caches. Each target has its own goroot cache..

for instance `tinygo info` and `tinygo info arudino-nan33` have to different caches. 

What is the best way to identify which code is going to be in a cache?

the cache for `tinygo info arudino-nano33` says it doesn't have certain modules.

```shell
$ joematpal/nano_33_test » tinygo info arduino-nano33
LLVM triple:       thumbv6m-unknown-unknown-eabi
GOOS:              linux
GOARCH:            arm
build tags:        cortexm baremetal linux arm atsamd21g18a atsamd21g18 atsamd21 sam arduino_nano33 tinygo math_big_pure_go gc.conservative scheduler.tasks serial.usb
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     /home/joseph/.cache/tinygo/goroot-e71eb3d56fb851142fc9d23380549fc0b3b7fda074f81c24b8887b5e9bdf40f5

error while importing tinygo.org/x/drivers/netlink: package net/netip is not in std (/home/joseph/.cache/tinygo/goroot-e71eb3d56fb851142fc9d23380549fc0b3b7fda074f81c24b8887b5e9bdf40f5/src/net/netip) 
```

deadprogram, 2024-09-08:
What is the output from running `tinygo env`?

sentientmachin3, 2024-10-15:
I'm having the same issue, but using a raspberry pico. 

I set `GOROOT` and `GOFLAGS` from the result of `tinygo info pico` command, run `tinygo build -o main.uf2 main.go` and I get the error `package sync/atomic is not in std `, but with a different path from the one I specified.

Apparently `tinygo` keeps creating new folders, so it never compiles. I guess I'm going to try with an earlier version.

sago35, 2024-10-15:
GOROOT and GOFLAGS are automatically set when running the tinygo command, so you don't need to configure them yourself. In general, it's better not to set them.

sentientmachin3, 2024-10-16:
ok good to know!

aykevl, 2024-10-16:
Indeed. Don't set `GOROOT` or `GOFLAGS`, they will only mess things up. And make sure they aren't set by checking `env` (run `env | grep GOROOT` or similar).

For IDEs, you can set `GOROOT` to the value in "cached GOROOT". For example, here is what the vscode-tinygo extension does:

https://github.com/tinygo-org/vscode-tinygo/blob/main/src/extension.ts#L51-L100

...but make sure these environment variables _only_ apply inside the IDE! The IDE should (hopefully) have a way to configure these environment variables.

kuchaguangjie, 2024-11-29:
I got the similar error, `go build` works well, `tinygo build` report: 
```
../../../../pkg/mod/github.com/gin-gonic/gin@v1.10.0/recovery.go:15:2: package net/http/httputil is not in std (/home/eric/.cache/tinygo/goroot-18fe219864c1abce0632092c6bb93c7ada89c1c5d45131f776258a833c68f038/src/net/http/httputil)
../../../../pkg/mod/github.com/gogf/gf@v1.16.9/net/ghttp/ghttp_server_pprof.go:10:2: package net/http/pprof is not in std (/home/eric/.cache/tinygo/goroot-18fe219864c1abce0632092c6bb93c7ada89c1c5d45131f776258a833c68f038/src/net/http/pprof)
../../../../pkg/mod/github.com/gogf/gf@v1.16.9/net/ghttp/internal/client/client.go:22:2: package net/http/cookiejar is not in std (/home/eric/.cache/tinygo/goroot-18fe219864c1abce0632092c6bb93c7ada89c1c5d45131f776258a833c68f038/src/net/http/cookiejar)
../../../../pkg/mod/golang.org/x/net@v0.25.0/http2/transport.go:26:2: package net/http/httptrace is not in std (/home/eric/.cache/tinygo/goroot-18fe219864c1abce0632092c6bb93c7ada89c1c5d45131f776258a833c68f038/src/net/http/httptrace)
```

kuchaguangjie, 2024-11-29:
I got this, tinygo support a sub set of go std api:
https://tinygo.org/docs/reference/lang-support/stdlib/

deadprogram, 2024-12-06:
Now closing since the issue appears to have been to changing the local config. Please reopen if needed. Thanks!

closed 2024-12-06 by deadprogram