tinygo-org/tinygo #4452: Add support for Raspberry Pi's RP2350 Dual Arm Cortex-M33

soypat, 2024-09-01:
This issue would seek to replicate [this PR done for the RP2040](https://github.com/tinygo-org/tinygo/pull/1897/files) for the new RP2350 chip.

- Page: https://www.raspberrypi.com/products/rp2350/
- Datasheet: https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

Support for RISCV Hazard3 cores is out of the scope of this issue.

deadprogram, 2024-09-01:
See https://github.com/cmsis-svd/cmsis-svd-data/pull/8 for one important part of this work to be done.

maxprofit, 2024-12-12:
Does any more work have to be done for this? It appears as if there have been a couple big merges, but I don't know the expected ending state of this issue?

deadprogram, 2025-02-04:
Now closing since it does appear to have been completed.

closed 2025-02-04 by deadprogram