tinygo-org/tinygo #4463: Failing build on Raspberry Pi 4

ShDDaniel, 2024-09-12:
Hello!

`Raspberry Pi 4 Model B Rev 1.4`
`tinygo version 0.33.0 linux/arm64 (using go version go1.23.0 and LLVM version 18.1.2)`

```
package main

import (
	"machine"
)

func main() {
	machine.I2C0.Configure(machine.I2CConfig{})
}
```

running `tinygo build` yields `/usr/local/lib/tinygo/src/machine/machine_generic.go:145: linker could not find symbol __tinygo_i2c_configure`, yet build with a target succeeds, e.g : `tinygo build -target=pico`

aykevl, 2024-09-12:
The machine package is not supported on Linux, that's why you see that error.

I agree the error could be much better though.

ShDDaniel, 2024-09-12:
Oh, that was unexpected. Indeed I wanted I2C on Raspberry Pi.
Is it possible that package will be supported in the future on Linux?

dgryski, 2024-09-19:
I think you probably want a package like https://periph.io/ which has support for using the hardware pins on a Raspberry Pi: https://periph.io/platform/raspberrypi/

closed 2024-09-19 by dgryski