tinygo-org/tinygo #4479: runtime should pass layout info to alloc() when possible

dgryski, 2024-09-23:
Right now we only have layout info for allocations created via the compiler.  We should make it possible for the runtime to pass layout info for common types at least ([]byte, []string, etc).

dgryski, 2024-09-23:
Tried a quick POC below
```
diff --git a/src/runtime/string.go b/src/runtime/string.go
index aeefe1d4..ceb5ff13 100644
--- a/src/runtime/string.go
+++ b/src/runtime/string.go
@@ -32,6 +32,12 @@ func stringEqual(x, y string) bool {
        return true
 }

+var (
+       // Constants for use with alloc()
+       gcLayoutZero   uintptr
+       gcLayoutNoPtrs = unsafe.Pointer(&gcLayoutZero)
+)
+
 // Return true iff x < y.
 //
 //go:nobounds
@@ -59,7 +65,7 @@ func stringConcat(x, y _string) _string {
                return x
        } else {
                length := x.length + y.length
-               buf := alloc(length, nil)
+               buf := alloc(length, gcLayoutNoPtrs)
                memcpy(buf, unsafe.Pointer(x.ptr), x.length)
                memcpy(unsafe.Add(buf, x.length), unsafe.Pointer(y.ptr), y.length)
                return _string{ptr: (*byte)(buf), length: length}
@@ -72,7 +78,7 @@ func stringFromBytes(x struct {
        len uintptr
        cap uintptr
 }) _string {
-       buf := alloc(x.len, nil)
+       buf := alloc(x.len, gcLayoutNoPtrs)
        memcpy(buf, unsafe.Pointer(x.ptr), x.len)
        return _string{ptr: (*byte)(buf), length: x.len}
 }
@@ -83,7 +89,7 @@ func stringToBytes(x _string) (slice struct {
        len uintptr
        cap uintptr
 }) {
-       buf := alloc(x.length, nil)
+       buf := alloc(x.length, gcLayoutNoPtrs)
        memcpy(buf, unsafe.Pointer(x.ptr), x.length)
        slice.ptr = (*byte)(buf)
        slice.len = x.length
@@ -100,7 +106,7 @@ func stringFromRunes(runeSlice []rune) (s _string) {
        }

        // Allocate memory for the string.
-       s.ptr = (*byte)(alloc(s.length, nil))
+       s.ptr = (*byte)(alloc(s.length, gcLayoutNoPtrs))

        // Encode runes to UTF-8 and store the resulting bytes in the string.
        index := uintptr(0)
```

but got an interp panic:

```
panic: interface conversion: interp.value is interp.literalValue, not interp.rawValue

goroutine 126 [running]:
github.com/tinygo-org/tinygo/interp.(*runner).readObjectLayout(0xc004032000, {0x1099fd7e8, 0xc004c24ee0})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/interp/memory.go:1295 +0x425
github.com/tinygo-org/tinygo/interp.(*runner).getLLVMTypeFromLayout(0xc004032000, {0x1099fd7e8, 0xc004c24ee0})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/interp/memory.go:1314 +0x37
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc004032000, 0xc003a28870, {0xc0011a3a90, 0x3, 0x0?}, 0xc00102cab0, {0xc004c24e78, 0x8})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/interp/interpreter.go:288 +0x3899
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc004032000, 0xc003a28820, {0x0, 0x0, 0x0?}, 0x0, {0x109895f4e, 0x4})
	/Users/dgryski/go/src/github.com/tinygo-org/tinygo/interp/interpreter.go:552 +0x821e

```

Seems like we'll either need to get the compiler to inject the pointer to the appropriate type layout, *or* figure out a way to get access to the existing symbol.  I'm concerned that fixing interp will actually move more of those string computations away from the interp phase and into runtime, which is sadly backwards from what we want.

dgryski, 2024-09-24:
Other requirements: we probably also need the ability to get a layout ptr given a reflect type, and possibly at compile time too (using some sort of compiler lowering pass from a fake method call).

dgryski, 2024-09-27:
Realized where I went wrong here; I should be able to create constant layouts for `[]byte` and `string` and pass those around; pointers only needed if the allocation doesn't fit in a word.

aykevl, 2024-09-27:
Correct, you can also look at what the compiler generates in an equivalent case:

https://github.com/tinygo-org/tinygo/blob/ef4f46f1d1550beb62324d750c496b2b4a7f76d0/compiler/testdata/gc.ll#L105

So for `stringConcat` for example (which allocates pointer-free memory) you can use something like `unsafe.Pointer(uintptr(0x3))`.

deadprogram, 2024-12-13:
Tagging to close on next release since #4495 seems to address this.

deadprogram, 2024-12-13:
Wait, my bad: this was already released. Now closing, thanks everyone!

closed 2024-12-13 by deadprogram