tinygo-org/tinygo #4480: ESP32 Can Bus (TWAI)

MrNavaStar, 2024-09-24:
The esp32 has a built in can controller than can interface with a TJA1050 or similar to allow for communication via the CAN protocol. As far as I can tell, this feature is not implemented and I haven't found any libraries that do so either.

Could someone point me in the direction of where to get started implementing this feature? Should it be added as to this repo as a board specific feature, or as a driver? 

Any help would be appreciated, thanks!

aykevl, 2024-09-28:
This should be implemented in the machine package. To get an idea of how it is done, you can see the I2C PR here: https://github.com/tinygo-org/tinygo/pull/4259

MrNavaStar, 2024-10-03:
I have tiny go setup, but I'm getting errors about "device/esp" not being present. How do you get this working? I can't find the module anywhere in the project

deadprogram, 2024-12-05:
@MrNavaStar did you happen to install TinyGo from source? If so, you probably need to run 

```shell
git submodule update --init
make gen-device
```

MrNavaStar, 2024-12-05:
Yeah I ended up figuring this out. I've currently been facing issues with neovim I believe. After generating devices, neovim seems to be unable to find many files that do exist, causing gopls to not work correctly. 

I'll try this again with something like goland when I can get my free license when the next semester starts.

deadprogram, 2024-12-06:
OK thanks everyone. Now closing.