tinygo-org/tinygo #4483: TinyGo support for SSD1306

ohchloeho, 2024-09-26:
I have tested both an Arduino Nano and ESP32 module with the SSD1306 OLED module while using the Go machine package with the tinygo SSD1306 driver ("tinygo.org/x/drivers/ssd1306"). It seems no matter how I flash TinyGo code to the modules the SSD1306 will not work. I'm certain this is a configuration issue as I'm able to flash LEDs and button controls to the pins. Please let me know if there's a resolution on this that I'm missing out on.

For reference, this is my code to print "Hello world!" on the OLED screen:

```
package main

import (
	"machine"
	"time"

	"tinygo.org/x/drivers/ssd1306"
)

func main() {
	// Initialize I2C
	machine.I2C0.Configure(machine.I2CConfig{
		SCL: machine.IO22, // Adjust for your ESP32 SCL pin
		SDA: machine.IO21, // Adjust for your ESP32 SDA pin
	})

	// Initialize the SSD1306 display
	display := ssd1306.NewI2C(machine.I2C0)
	display.Configure(ssd1306.Config{
		Width:  128,
		Height: 64,
	})

	// Clear the display buffer
	display.ClearBuffer()

	drawText(display, 0, 0, "Hello World!")
	display.Display()

	for {
		time.Sleep(time.Second)
	}
}

func drawText(display *ssd1306.Device, x int16, y int16, text string) {
	for i := int16(0); i < int16(len(text)); i++ {
		display.SetPixel(x+i*6, y, 1) 
	}
}

```

aykevl, 2024-09-26:
For the ESP32 you need to specify a board, not just `-target=esp32`.

I don't know what's up with the Arduino Nano, does it produce the same error message?

ohchloeho, 2024-09-30:
Thank you for replying! i'm using the ESP-WROOM-32 board, it's not listed within TinyGo and I've tried flashing with the other options.

Can you guide me through how to do this?

deadprogram, 2024-12-02:
The import

```go
import (
    "github.com/tinygo-org/tinygo/src/machine"
)
```

Is not valid. It should be only:

```
import (
    "machine"
)
```

deadprogram, 2025-02-08:
Closing since question appears to have been answered. Please reopen if needed.

closed 2025-02-08 by deadprogram