tinygo-org/tinygo #4484: package net/http/cookiejar is not in std

fardeadok, 2024-09-30:
how to fix?

fardeadok, 2024-09-30:
lol https://tinygo.org/docs/reference/lang-support/stdlib/#nethttp

net/http
This package cannot be imported because the following dependencies cannot be compiled:

[net/http/httptrace](https://tinygo.org/docs/reference/lang-support/stdlib/#nethttphttptrace)

tinygo dont support net?)))

M09Ic, 2025-07-04:
I have the same problem.
```
tinygo build .
go: downloading github.com/chainreactors/logs v0.0.0-20241115105204-6132e39f5261
go: downloading github.com/jessevdk/go-flags v1.6.1
go: downloading github.com/chainreactors/fingers v1.0.2-0.20250512141956-7380518a85e9
go: downloading github.com/chainreactors/neutron v0.0.0-20250113061151-32771f256b1f
go: downloading github.com/chainreactors/parsers v0.0.0-20250605044448-6bc270f12c0e
go: downloading github.com/chainreactors/proxyclient v1.0.3
go: downloading github.com/chainreactors/utils v0.0.0-20250624171356-667aa721dae0
go: downloading github.com/panjf2000/ants/v2 v2.9.1
go: downloading golang.org/x/net v0.0.0-20200202094626-16171245cfb2
go: downloading golang.org/x/sys v0.0.0-20200223170610-d5e6a3e2c0ae
go: downloading github.com/chainreactors/files v0.0.0-20240716182835-7884ee1e77f0
go: downloading github.com/facebookincubator/nvdtools v0.1.5
go: downloading gopkg.in/yaml.v3 v3.0.1
go: downloading github.com/M09ic/go-ntlmssp v0.0.0-20230312133735-dcccd454dfe0
go: downloading golang.org/x/text v0.3.3
go: downloading github.com/go-dedup/simhash v0.0.0-20170904020510-9ecaca7b509c
go: downloading github.com/twmb/murmur3 v1.1.8
go: downloading github.com/shadowsocks/go-shadowsocks2 v0.1.5
go: downloading golang.org/x/crypto v0.0.0-20191205180655-e7c4368fe9dd
go: downloading github.com/mozillazg/go-pinyin v0.20.0
go: downloading github.com/Knetic/govaluate v3.0.0+incompatible
go: downloading github.com/davecgh/go-spew v1.1.1
go: downloading github.com/weppos/publicsuffix-go v0.15.1-0.20220329081811-9a40b608a236
go: downloading github.com/asaskevich/govalidator v0.0.0-20230301143203-a9d515a09cc2
go: downloading github.com/hashicorp/go-version v1.6.0
go: downloading github.com/spaolacci/murmur3 v1.1.0
go: downloading github.com/go-dedup/text v0.0.0-20170907015346-8bb1b95e3cb7
go: downloading github.com/go-dedup/megophone v0.0.0-20170830025436-f01be21026f5
go: downloading github.com/riobard/go-bloom v0.0.0-20200614022211-cdc8013cb5b3
../../go/pkg/mod/github.com/weppos/publicsuffix-go@v0.15.1-0.20220329081811-9a40b608a236/publicsuffix/publicsuffix.go:12:2: package net/http/cookiejar is not in GOROOT (/root/.cache/tinygo/goroot-1ede9f467477d49c6086e4b8224f85420f11733969ce4f3fe9efcc8a3e12c4d6/src/net/http/cookiejar)
```

deadprogram, 2025-08-10:
`httptrace` was just added in https://github.com/tinygo-org/net/pull/39

deadprogram, 2025-08-10:
Also see https://github.com/tinygo-org/tinygo/pull/4990

0pcom, 2026-10-20:
This is still the case on dev. A program that imports net/http/cookiejar fails with "package net/http/cookiejar is not in std". net/http is overridden as a whole, so Go's subdirectories never reach the merged goroot. This branch merges the directory and pins only the subdirectories TinyGo replaces: https://github.com/tinygo-org/tinygo/compare/dev...0magnet:tinygo:loader-net-http-subdirs. With it the program builds and runs.