tinygo-org/tinygo #4485: Go compatibility matrix is out of date

inliquid, 2024-10-01:
There is very good document [Go compatibility matrix](https://tinygo.org/docs/reference/go-compat-matrix/) which describes mapping between supported big Go and `tinygo` versions, but unfortunately outdated, the last Go version mentioned there is `go1.20`.

I have discovered this when was investigating some issues in related Wasm projects. I think it makes sense to update, and maybe add some automation here? Not sure about that last point.

calebschoepp, 2024-10-29:
I would also find an up to date version of this beneficial.

davecheney, 2026-10-27:
Thanks for flagging this. The Go compatibility matrix has since been updated through Go 1.27. Its `0.42.0+` row also covers TinyGo 0.43 and 0.44.

Closing as resolved. Please feel free to reopen if you spot any remaining omissions.

closed 2026-10-27 by davecheney