tinygo-org/tinygo #4486: Compiler crash on missing function body via interface with type parameter

jmillikin, 2024-10-02:
```go
package main

type Message[T any] interface {
	isMessage(T)
}

type IsGeneratedMessage[T any] struct{}

func (IsGeneratedMessage[T]) isMessage(T)

type HelloMsg struct {
	IsGeneratedMessage[HelloMsg]
}

// package main
func printSize[T any](message Message[T]) {
}

func main() {
	printSize(&HelloMsg{})
}
```

```
% go version
go version go1.23.1 darwin/arm64
% go build -o main
# example.com/example
./main.go:9:6: missing function body
```

```
% tinygo version
tinygo version 0.33.0 darwin/arm64 (using go version go1.23.1 and LLVM version 18.1.2)
% tinygo build -o main         
panic: runtime error: index out of range [0] with length 0

goroutine 82 [running]:
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionStart(0x14000022900, 0x0)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1207 +0xdf0
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0x14000022900)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1296 +0x2c
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0x14000194000, 0x14000896400)
	/Users/runner/work/tinygo/tinygo/compiler/symbol.go:224 +0xc94
github.com/tinygo-org/tinygo/compiler.(*builder).createFunctionCall(0x14000022780, 0x1400362d340)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1870 +0x978
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0x14000022780, {0x1088b7900, 0x1400362d300})
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:2035 +0x1ebc
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0x14000022780, {0x1088b77e0, 0x1400362d300})
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1448 +0x898
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0x14000022780)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1337 +0x6a8
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getFunction(0x14000194000, 0x14000896800)
	/Users/runner/work/tinygo/tinygo/compiler/symbol.go:224 +0xc94
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getTypeMethodSet(0x14000194000, {0x1088b3f08, 0x140035b8b10})
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:621 +0x1d4
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getTypeCode(0x14000194000, {0x1088b3f08, 0x140035b8b10})
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:420 +0x2bdc
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getTypeCode(0x14000194000, {0x1088b3ee0, 0x140046f08d0})
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:355 +0x24a0
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getTypeCode(0x14000194000, {0x1088b3c60, 0x140000f64d0})
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:295 +0x2300
github.com/tinygo-org/tinygo/compiler.(*compilerContext).getTypeCode(0x14000194000, {0x1088b3f08, 0x1400358a7e0})
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:325 +0x29b0
github.com/tinygo-org/tinygo/compiler.(*builder).createMakeInterface(0x14000022600, {0x1088b7870?}, {0x1088b3f08, 0x1400358a7e0}, 0x2?)
	/Users/runner/work/tinygo/tinygo/compiler/interface.go:80 +0x6c
github.com/tinygo-org/tinygo/compiler.(*builder).createExpr(0x14000022600, {0x1088b76c0, 0x14003df3a40})
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:2219 +0x1890
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0x14000022600, {0x1088b87a0, 0x14003df3a40})
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1448 +0x898
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0x14000022600)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:1337 +0x6a8
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0x14000194000, {0x14000175c00?}, 0x14003c23880)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:853 +0x3a4
github.com/tinygo-org/tinygo/compiler.CompilePackage({0x1400344d8f0?, 0x61?}, 0x1400347ba40, 0x14003c23880, {0x10111b6ffada033c?}, 0x1070d2913?, 0x68?)
	/Users/runner/work/tinygo/tinygo/compiler/compiler.go:312 +0x394
github.com/tinygo-org/tinygo/builder.Build.func3(0x140049d2cc0)
	/Users/runner/work/tinygo/tinygo/builder/build.go:375 +0x18c
github.com/tinygo-org/tinygo/builder.runJob(0x140049d2cc0, 0x140048c6000)
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:222 +0x48
created by github.com/tinygo-org/tinygo/builder.runJobs in goroutine 1
	/Users/runner/work/tinygo/tinygo/builder/jobs.go:123 +0x43c
```

deadprogram, 2024-12-13:
@jmillikin so the issue you are reporting is that the error message is not the same, since the code does not compile in big Go either?

jmillikin, 2024-12-14:
My general assumption is that a crash (in this case, due to array index bounds checking) indicates some sort of serious logic bug elsewhere in the program. So even though the code's behavior could be summarized as "writes some text to stderr and then exits", I tend to think of it as categorically different from an incorrect/unclear error message.

If you don't consider this to be a bug then feel free to ignore/close.

aykevl, 2024-12-14:
Yes this is a bug, I have a fix and I'll make a PR.

aykevl, 2024-12-14:
Here is the fix: https://github.com/tinygo-org/tinygo/pull/4657

Thank you for reporting!

deadprogram, 2024-12-14:
Tagged to close this issue after next release.

QuLogic, 2025-01-26:
I'm reopening this issue because of https://github.com/tinygo-org/tinygo/pull/4657#issuecomment-2566343500