tinygo-org/tinygo #4487: `reflect.Value` methods `Seq` and `Seq2` missing

jmillikin, 2024-10-02:
These two methods are new in Go 1.23 as part of the new iterators feature:
- https://pkg.go.dev/reflect#Value.Seq
- https://pkg.go.dev/reflect#Value.Seq2

```
% cat main.go 
package main

import "reflect"

func main() {
	var v reflect.Value
	_ = v.Seq
	_ = v.Seq2
}
% go build -o main
% tinygo build -o main
# example.com/example
main.go:7:8: v.Seq undefined (type reflect.Value has no field or method Seq)
main.go:8:8: v.Seq2 undefined (type reflect.Value has no field or method Seq2)
```

dgryski, 2024-10-02:
Implementing this means pulling in https://github.com/golang/go/blob/master/src/reflect/iter.go and https://github.com/golang/go/tree/master/src/iter

dxps, 2025-03-12:
Another example of the issue (using tinygo 0.36.0 and Go 1.24.1 on on linux/amd64):
```shell
❯ GOARCH=wasm GOOS=js tinygo build -o test.wasm ./cmd/gdm_pwa/main.go 
# text/template
.../go/src/text/template/exec.go:405:22: val.Seq undefined (type reflect.Value has no field or method Seq)
.../go/src/text/template/exec.go:454:17: val.Type().CanSeq undefined (type reflect.Type has no field or method CanSeq)
.../go/src/text/template/exec.go:460:23: val.Seq undefined (type reflect.Value has no field or method Seq)
.../go/src/text/template/exec.go:471:17: val.Type().CanSeq2 undefined (type reflect.Type has no field or method CanSeq2)
.../go/src/text/template/exec.go:473:26: val.Seq2 undefined (type reflect.Value has no field or method Seq2)
❯ 
```