tinygo-org/tinygo #4489: [CGO] Invalid behaviour on function call

Leopotam, 2024-10-02:
```h
// native.h
typedef struct Vector2
{
    float x;
    float y;
} Vector2;

Vector2 Vector2Add(Vector2 v1, Vector2 v2)
{
    Vector2 result = {v1.x + v2.x, v1.y + v2.y};
    return result;
}
```
```go
// main.go
package main

// #include "native.h"
import "C"
import "fmt"

func main() {
	a := C.Vector2{1, 2}
	b := C.Vector2{1, 2}
	c := C.Vector2Add(a, b)
	fmt.Println(c)
}
```
```sh
> go run main.go
{2 4}
```
```sh
> tinygo run main.go
{3 2}
```
```sh
> tinygo version
tinygo version 0.33.0 darwin/amd64 (using go version go1.23.2 and LLVM version 18.1.2)
```

supremestranger, 2024-10-02:
Windows:
```sh
> tinygo.exe run main.go
{0 0}
```

```sh
> go run main.go
{2 4}
```

dgryski, 2024-10-03:
Ugh, this has got to be calling convention differences for non-integer-sized objects.

Leopotam, 2024-10-03:
Oh, cgo in tinygo supports only integer-sized objects - its not issue, nevermind then.

dgryski, 2024-10-03:
Reopening because it's still a cgo bug/limitation.

Leopotam, 2024-10-19:
@dgryski any workaround for this issue? Can I compile c-code as part of go-project to wasm-browser target with tinygo? Standard go-compiler doesnt support it.