tinygo-org/tinygo #4491: wasip1 `syscall` package needs `syscall.FDFLAG_NONBLOCK`

dgryski, 2024-10-02:
Go's `internal/syscall/unix` package expects it, and is needed for some tests

```
~ $ tinygo test -target=wasip1 mime/multipart
FAIL	mime/multipart	0.000s
# internal/syscall/unix
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:19:23: undefined: syscall.FDFLAG_NONBLOCK
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:23:22: undefined: syscall.FDFLAG_NONBLOCK
~ $ tinygo test -target=wasip1 net/mail
FAIL	net/mail	0.000s
# internal/syscall/unix
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:19:23: undefined: syscall.FDFLAG_NONBLOCK
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:23:22: undefined: syscall.FDFLAG_NONBLOCK
~ $ tinygo test -target=wasip1 net/textproto
FAIL	net/textproto	0.000s
# internal/syscall/unix
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:19:23: undefined: syscall.FDFLAG_NONBLOCK
/Users/dgryski/go/src/go.googlesource.com/go/src/internal/syscall/unix/nonblocking_wasip1.go:23:22: undefined: syscall.FDFLAG_NONBLOCK
```

aykevl, 2024-10-02:
I was actually changing some targets over to use the Go stdlib syscall package instead. That would probably solve this issue.

deadprogram, 2024-12-06:
@dgryski @aykevl was this addressed by the changes to use Go stdlib package?

deadprogram, 2025-03-16:
Verified that this is now working:

```shell
$ tinygo test -target=wasip1 mime/multipart
ok      mime/multipart  25.250s
```

Now closing, thanks everyone!

closed 2025-03-16 by deadprogram

dgryski, 2025-03-16:
We can probably add `mime/multipart` to the stdlib packages list to test on WASI in `GNUMakefile`.