tinygo-org/tinygo #4507: runtime.NumGoRoutine lies

dgryski, 2024-10-07:
We should fix https://github.com/tinygo-org/tinygo/blob/release/src/runtime/debug.go#L17

aykevl, 2024-10-07:
The runtime currently does not track which goroutines exist. So this information is not currently available.

This actually has an interesting benefit: goroutines are just plain old data structures to the garbage collector. If two goroutines are deadlocked with each other for example, they should get garbage collected like any other object on the heap. Because the GC doesn't know these are goroutines, it can't lower a count of goroutines for example.

(This could be fixable in the precise GC, by using special values for the "layout" value).

deadprogram, 2026-10-10:
This is fixed in 0.43.0. `runtime.NumGoroutine` now returns the real number of goroutines, added in 38222384.

Closing, but please reopen if you still see a wrong count.

closed 2026-10-10 by deadprogram