tinygo-org/tinygo #4512: compiler: ensure iterators are inlined in -opt=z

eliasnaur, 2024-10-09:
```go
$ cat iter_test.go
package inlineiter

import (
	"iter"
	"testing"
)

var slice = make([]int, 1000)

func TestIterAllocs(t *testing.T) {
	tests := []struct {
		name string
		body func() int
	}{
		{
			"for-range-func",
			func() int {
				sum := 0
				for v := range Iterator([]int{0, 1, 2}) {
					sum += v
				}
				return sum
			},
		},
		{
			"callback-func",
			func() int {
				sum := 0
				Iterator([]int{0, 1, 2})(func(v int) bool {
					sum += v
					return true
				})
				return sum
			},
		},
	}
	for _, test := range tests {
		t.Run(test.name, func(t *testing.T) {
			allocs := testing.Benchmark(func(b *testing.B) {
				for range b.N {
					sum := test.body()
					if sum != 3 {
						t.Errorf("sum %d, want 3", sum)
					}
				}
			}).AllocsPerOp()
			if allocs > 0 {
				t.Errorf("Layout allocates %d, expected %d", allocs, 0)
			}
		})
	}
}

func Iterator(ints []int) iter.Seq[int] {
	return func(yield func(int) bool) {
		for _, v := range ints {
			if !yield(v) {
				break
			}
		}
	}
}
$ go test
PASS
ok  	seedhammer.com/inlineiter	4.616s
$ tinygo test
--- FAIL: TestIterAllocs (3.08s)
    --- FAIL: TestIterAllocs/for-range-func (1.58s)
        Layout allocates 5, expected 0
    --- FAIL: TestIterAllocs/callback-func (1.50s)
        Layout allocates 3, expected 0
FAIL
FAIL	seedhammer.com/inlineiter	3.297s
$ tinygo test -opt 2
ok  	seedhammer.com/inlineiter	3.211s
```

The idiom for iterator constructors is to return a closure that captures the iterator arguments, if any. It's crucial for the performance of range-over-func that the constructor doesn't incur allocations. Therefore, it seems to me functions that (nearly) immediately return a closure should always be inlined, regardless of optimization level. Or, at the very least, any function that returns a function that can be ranged over.

dgryski, 2024-10-21:
Probably we'll need to figure out exactly which optimization pass(es) are responsible for this particular optimization and hope that turning just those ones on is sufficient to make this "always work".

aykevl, 2024-10-22:
I did some more investigation and it looks like everything is inlined as expected (even at `-opt=z`), but there's indeed one allocation left. What I suspect is happening is that while this allocation could be optimized away, the inlining happens _after_ our heap-to-stack optimization pass.

I think the right way to fix this is by using the heap-to-stack transform that's part of the LLVM Attributor pass. Unfortunately that pass doesn't know that pointers returned by our `runtime.alloc` pass will never explicitly be freed. That means the heap-to-stack transform in the Attributor pass isn't as effective as it could be. I might write a patch to LLVM to fix this.

aykevl, 2024-10-22:
Wrote a patch: https://github.com/llvm/llvm-project/pull/113299

eliasnaur, 2024-10-23:
> While I agree it would be great to fix this, in general you can't really rely on the compiler optimizing anything (including heap allocs).
> 
> Also note that `-opt=z` (the default) optimizes for code size, and not much else. So inlining would have to reduce or at least not increase code size. If you want to optimize for performance, you need to explicitly use `-opt=2`.

"Performance" is actually not even that important for me. The deal breaker for my uses is the tendency of (typically small) frequent allocations to fragment the heap so much that out-of-memory panics occur when a larger allocation arrives. My current strategy is to eliminate allocations for frequently run code (GUI refreshes) which both avoids fragmentation and is faster.

A longer term strategy is to not fragment the heap so much, but as you wrote somewhere a moving collector is difficult and won't help performance.

Thank you for the LLVM patch. Do I have to do anything special on the TinyGo side to try it out, other than patching LLVM and adding the "nofree" attribute? Also, is the comment "for your own purpose, you can disable/filter out H2S AA explicitly when you construct the attributor" relevant to achieve better H2S in TinyGo?

eliasnaur, 2024-11-03:
Can anything be done on our side while we wait for a LLVM solution? For example, will marking every pointer argument to every function "nofree" make the H2S pass effective?

davecheney, 2026-09-21:
This still reproduces for me with TinyGo 0.43.0-dev-bc7470b2 (Go 1.27.1, LLVM 20.1.8, darwin/arm64 host), and I found one important wrinkle: `-opt=2` does not eliminate the allocations when the iterator is the stdlib `slices.Backward` (`iter.Seq2`).

aykevl previously wrote: “If you want to optimize for performance, you need to explicitly use `-opt=2`.” That appears to be sufficient for the hand-written `iter.Seq[int]` reproducer, but not for this stdlib case. Building the iterator version of a PDP-11 emulator at all three optimization levels produced byte-identical `-print-allocs='.'` output:

```
-opt=1   cpu.go:494, 495, 496  and  cpu.go:546, 547  -> allocations
-opt=2   identical
-opt=z   identical
```

The command was:

```
tinygo build -o tmp/t.uf2 -opt=$O -target=waveshare-rp2350-pizero -scheduler=cores \
  -tags model.1145 -print-allocs='.' ./cmd/pidp11-bootmon
```

The relevant code was three loops of this form on the interrupt path:

```go
for i, irq := range slices.Backward(cpu.interruptQueue) {
```

Rewriting them as plain reverse index loops:

```go
for i := len(cpu.interruptQueue) - 1; i >= 0; i-- {
    irq := cpu.interruptQueue[i]
```

removed the cost in a real firmware run on a Waveshare RP2350-PiZero (RP2350/Cortex-M33): the iterator version caused 136,158 allocations/sec and 13 GC cycles/sec, while the index-loop version measured 3 allocations/sec and 0 GC cycles/sec (the residual 3/sec is from a debug reporter’s `fmt.Fprintf` boxing; the emulator itself was effectively zero-allocation). `interruptWaitRelease` is called once per emulated WAIT instruction, roughly 29k times a second at the monitor prompt, so this is about 3–4 heap objects per call. The independent runtime measurement is also evidence that these are real allocations, rather than only `-print-allocs` artefacts.

We have a straightforward workaround, so this is not a request for a fix on any particular timeline. I wanted to report the `slices.Backward`/`iter.Seq2` distinction because the documented `-opt=2` workaround is not sufficient here and could otherwise send the next person looking in the wrong place.